ckb-next/ckb-next #927: Change volume scrolwheel values

Lowrida, 2023-02-07:
### Describe your situation in detail.

How can i change the values of the volume scroll wheel? At the moment it changes at steps of 6, but i want to set it at smaller steps.

Device is a K70 MK2

### Additional Details

Add any additional details here.

### Generated Report (if applicable)

Drag your generated report here if it's applicable.

Lowrida, 2023-02-09:
I guess it can't..

tatokis, 2023-02-11:
All ckb-next does is pass events from the keyboard to the rest of your OS. It works on a much lower level and thus can not handle this. It's up to your desktop environment to configure the volume steps.

Alternatively, you could remap the volume wheel to a script that manually changes the volume in whatever steps you desire.

closed 2023-02-11 by tatokis

Lowrida, 2023-02-13:
With my previous keyboard, Cooler Master CK352, it did smaller steps. Same OS install (Mint 20.3).
When i scroll with the mouse over the sound applet it takes steps of 2. I don't understand were i can change it for the keyboard? Is this set in a configuration file somewere?

fleischie, 2023-02-14:
This forum thread seems to indicate, thay media key (i.e. XF86AudioRaiseVolume, XF86AudioLowerVolume) steps in cinnamon are hardcoded to 5%: https://forums.linuxmint.com/viewtopic.php?t=312162 Maybe it also contains additional information for you to.implement your desired behavior.

Scrolling the mouse over an applet's slider works differently, as it does not use these media key events.

Lowrida, 2023-02-14:
I saw that post too, but it seems a lot of work and installing extra software for just a small change.. but thanks.

tatokis, 2023-02-15:
You can bind scroll up and scroll down to something like

```
pactl -- set-sink-volume @DEFAULT_SINK@ +2%
```
and
```
pactl -- set-sink-volume @DEFAULT_SINK@ -2%
```

Although keep in mind you won't get any visual or audio feedback when changing the volume.

Lowrida, 2023-02-15:
My sound is set to 5.1, but most of the time the sub and rear speakers are set to 0. When i use the applet or volume scroll wheel only the channels that are not set to 0 change.
I have tried that line but the problem is that it ups and downs all channels. Even the channels that are temporary set to 0.

Lowrida, 2023-03-04:
I made 2 scripts that do the trick. Maybe someone else finds it useful?
 

volume-up.sh
```
#!/bin/bash
########################
#  Keyboard Volume UP  #
########################
#
# Volume steps
steps="2"
#
# Path to sound clip (leave empty if no sound is needed)
sound="/usr/share/mint-artwork/sounds/volume.oga"
#
#
#
##################################################################################################
############################### Do not change after this line ####################################
##################################################################################################

sink=$(pulsemixer -l | grep Default | grep Sink | awk -F" " '{print $3}' | awk -F"," '{print $1}')
vols=$(pulsemixer --id $sink --get-volume | awk -F" " '{printf $1"\n"$2"\n"$3"\n"$4"\n"$5"\n"$6}')
maxvol=$(gsettings get org.cinnamon.desktop.sound maximum-volume)
x=0

while read line;
do

  if [ "$line" -lt "$maxvol" ]; then
      x=$((x+1))
.
      if ! [ "$line" = "0" ]; then
        up=$((line+steps))
        channels="$channels $up"
      else
        channels="$channels 0"
      fi

      if [ "$x" = "1" ]; then
        if [ "$line" -lt 100 ]; then
           volume=""
        else
           volume="-v 0"
        fi
      fi
  else
    exit   # If max volume is reached, then exit
  fi

done < <(echo "$vols")

channels=${channels:1}.
channels=$(sed 's/ /:/g' <<<$channels)
 if [ "$channels" = "0:0:0:0:0:0" ]; then channels="10:10:0:0:0:0"; fi

pulsemixer --id $sink --set-volume-all "$channels" &

play --buffer 256 -q $volume "$sound" pitch 50 speed 1 &>/dev/null
```
 

volume-down.sh
```
#!/bin/bash
##########################
#  Keyboard Volume DOWN  #
##########################
#
# Volume steps
steps="2"
#
# Path to sound clip (leave empty if no sound is needed)
sound="/usr/share/mint-artwork/sounds/volume.oga"
#
#
#
##################################################################################################
############################### Do not change after this line ####################################
##################################################################################################

sink=$(pulsemixer -l | grep Default | grep Sink | awk -F" " '{print $3}' | awk -F"," '{print $1}')
vols=$(pulsemixer --id $sink --get-volume | awk -F" " '{printf $1"\n"$2"\n"$3"\n"$4"\n"$5"\n"$6}')
x=0

while read line;
do

     x=$((x+1))

     if ! [ "$line" = "0" ]; then
       up=$((line-steps))
       channels="$channels $up"
     else
       channels="$channels 0"
     fi

     if [ "$x" = "1" ]; then
       if [ "$line" -lt 100 ]; then
          volume=""
       else
          volume="-v 0"
       fi
    fi

done < <(echo "$vols")

channels=${channels:1}.
channels=$(sed 's/ /:/g' <<<$channels)

pulsemixer --id $sink --set-volume-all "$channels" &

play --buffer 256 -q $volume "$sound" pitch -50 speed 1 &>/dev/null
```