ckb-next/ckb-next #929: firmware update failure

MooCrumpus, 2023-02-12:
### Affected Devices

K95 RGP Platinum

### Linux Distribution

Archlinux

### Installation Method

aur

### ckb-next version

ckb-next 0.5.0

### What did you expect?

Searching for firmware update should proceed. Instead it reports
center>Es gab ein Problem beim Abrufen des Status für dieses Gerät.
Möchten Sie manuell eine Datei auswählen?

Translates to:
center>There was a problem retrieving the status for this device.
Do you want to select a file manually?

### Did you try anything that may have helped?

used iCue on windows (qemu).
[image]

### Additional Details

Add any additional details here.

### Generated Report

Drag your generated report here.

Fabick75, 2023-06-24:
Hi, I've the same problem in the latest release 0.60 with my Sabre mouse and K70 LUX RGB keyboard. There seems that no usefull log was generated to give you hints about the problem. My system is a linux Slackware 64 bit, with kernel version 6.1.35.
I've downloaded the git repository and compiled as usual with quickinstall command.
If someone have idea about to resolve this, please update the thread.

gravityfargo, 2024-07-12:
Hardware Playback is not supported for your device. #95

closed 2024-07-12 by gravityfargo