ckb-next/ckb-next #931: K55: winlock led does not toggle on

LockBlock-dev, 2023-02-18:
### Affected Devices

K55 RGB

### Linux Distribution

Arch

### Installation Method

AUR (ckb-next-git)

### ckb-next version

0.5.0

### What did you expect?

When the winlock button is pressed, the non-RGB LED should be on.

### Did you try anything that may have helped?

I captured the winlock USB packet on Windows with Wireshark and it uses the same as the K68: `07 05 09`.
Knowing this, I tried adding a 2nd OR (`||`) for the `P_K55` keyboard in this if statement (made for K66 and K68) without success: https://github.com/ckb-next/ckb-next/blob/c7d04173cc8eb7fa1b6f94ff1a4496a2590be0bc/src/daemon/led_keyboard.c#L117-L127
However, setting the `winlock_status` to `1` forces the winlock led to be enabled. I'm guessing the problem comes from the daemon not sending the correct information in the new lightmap. This may be because the K55 uses a 3-zones RGB mode and is not fully addressable.

Changing https://github.com/ckb-next/ckb-next/blob/6c5744262a194435ae0f386c430ed780522d1243/src/daemon/keymap.c#L420 to
```
    { "lock",          114, KEY_CORSAIR },
```
doesn't work either.

### Additional Details

Add any additional details here.

### Generated Report

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/10775126/ckb-next-dev-detect-report.gz)
The report might be a mess i restarted the daemon a lot to test my edits. There aren't any error as far as I know.

LockBlock-dev, 2023-02-18:
This might be related to #868.

LockBlock-dev, 2023-02-19:
After hours of work I was able to get something working, you can see it [here](https://github.com/LockBlock-dev/ckb-next/commit/b891dada0d77c609dc18bf8f121b3362eaa9d9b0). I guess it would be a problem in case someone would remap the lwin/rwin keys though.

tatokis, 2023-02-26:
It is not related to https://github.com/ckb-next/ckb-next/issues/868 as that keyboard uses a completely different protocol and the functions are not shared. Same reason why changing 1 to 114 in that line didn't do anything as it is a different keymap.

The way this works is: you press winlock, the daemon tells the GUI that you pressed winlock, and the GUI then tells the daemon to turn the light on or off (with an rgb command to the pipe). The patch you created is obviously not upstreamable, and you shouldn't have to do anything other than add your K55 to that if statement you identified in led_keyboard.c.

I don't see any keymap patches for the K55, so the "lock" key still lives at index 8. This probably means that the GUI for some reason isn't sending winlock correctly. Make sure the GUI in the performance tab is set to actually enable the winlock led.

Try attaching strace to `ckb-next`, create a new profile without animations to stop the spam, and see what values get sent (if any) when you press the winlock key. You can do that with ``sudo strace -p `pidof ckb-next` -f -e write -s 999999``.
Also, try printing the individual values of `newlight->r[8] || newlight->g[8] || newlight->b[8]` in that function.

tatokis, 2023-02-27:
I just had a thought. It's possible the GUI isn't sending the winlock led data because the K55 keymap in the GUI doesn't have a "lock" light. Try adding one manually in keymap.cpp and see if that does anything.

LockBlock-dev, 2023-03-04:
Thanks for your quick response!

I know my patch is dirty and is not upstreamable I only made it to fit my needs.

I do have the `Indicate Windows Lock` option enabled in the `Performance` tab. Values are default `#ffffff` and `#000000` with alpha channel of `255`:
[image]

`strace` is giving me this output when i press the <kbd>winlock</kbd> key. I can clearly see the `unbind lwin rwin` command being sent to the daemon:
<details>
<summary>enabling lock:</summary>
<code>[pid  2304] write(4, "\1\0\0\0\0\0\0\0", 8) = 8
[pid  2304] write(42, "\n@1 rebind all notify all bind rbrace:rbrace bind home:home bind b:b bind fn:fn bind t:t bind 8:8 bind rshift:rshift bind f7:f7 bind ralt:ralt bind f10:f10 bind 5:5 bind num3:num3 bind num6:num6 bind rmenu:rmenu bind f2:f2 bind enter:enter bind 2:2 bind quote:quote bind equal:equal bind f1:f1 bind num2:num2 bind w:w bind slash:slash bind x:x bind zone3:zone3 bind f8:f8 bind q:q bind 0:0 bind left:left bind numplus:numplus bind next:next bind numslash:numslash bind grave:grave bind tab:tab bind f3:f3 bind d:d unbind g4 bind hash:hash bind esc:esc bind num8:num8 bind pgdn:pgdn bind f:f bind numstar:numstar bind 1:1 bind comma:comma bind num1:num1 bind f12:f12 bind zone2:zone2 bind z:z bind numdot:numdot bind j:j unbind g2 bind lalt:lalt bind bslash_iso:bslash_iso bind c:c bind ins:ins bind f11:f11 bind num0:num0 bind right:right bind 4:4 unbind light bind end:end bind volup:volup unbind g1 bind numminus:numminus bind 7:7 bind l:l bind minus:minus bind 3:3 bind down:down unbind g5 bind v:v bind i:i bind 9:9 bind f9:f9 bind y:y bind lbrace:lbrace bind s:s bind pause:pause bind stop:stop bind bspace:bspace bind a:a bind numenter:numenter bind up:up bind n:n bind e:e bind num7:num7 bind voldn:voldn bind r:r bind num9:num9 bind u:u bind num4:num4 unbind mr bind caps:caps bind dot:dot bind p:p unbind g6 bind scroll:scroll bind play:play bind f4:f4 bind prtscn:prtscn bind pgup:pgup bind o:o bind f5:f5 unbind lock bind numlock:numlock bind space:space bind m:m bind rctrl:rctrl bind k:k bind lctrl:lctrl bind zone1:zone1 unbind g3 bind colon:colon bind lshift:lshift bind lwin:lwin bind g:g bind 6:6 bind h:h bind f6:f6 bind mute:mute bind del:del bind prev:prev bind num5:num5 unbind lwin rwin\nmacro clear\n\n", 1722) = 1722</code>
</details>
<details>
<summary>disabling lock:</summary>
<code>[pid  2304] write(4, "\1\0\0\0\0\0\0\0", 8) = 8
[pid  2304] write(42, "\n@1 rebind all notify all bind rbrace:rbrace bind home:home bind b:b bind fn:fn bind t:t bind 8:8 bind rshift:rshift bind f7:f7 bind ralt:ralt bind f10:f10 bind 5:5 bind num3:num3 bind num6:num6 bind rmenu:rmenu bind f2:f2 bind enter:enter bind 2:2 bind quote:quote bind equal:equal bind f1:f1 bind num2:num2 bind w:w bind slash:slash bind x:x bind zone3:zone3 bind f8:f8 bind q:q bind 0:0 bind left:left bind numplus:numplus bind next:next bind numslash:numslash bind grave:grave bind tab:tab bind f3:f3 bind d:d unbind g4 bind hash:hash bind esc:esc bind num8:num8 bind pgdn:pgdn bind f:f bind numstar:numstar bind 1:1 bind comma:comma bind num1:num1 bind f12:f12 bind zone2:zone2 bind z:z bind numdot:numdot bind j:j unbind g2 bind lalt:lalt bind bslash_iso:bslash_iso bind c:c bind ins:ins bind f11:f11 bind num0:num0 bind right:right bind 4:4 unbind light bind end:end bind volup:volup unbind g1 bind numminus:numminus bind 7:7 bind l:l bind minus:minus bind 3:3 bind down:down unbind g5 bind v:v bind i:i bind 9:9 bind f9:f9 bind y:y bind lbrace:lbrace bind s:s bind pause:pause bind stop:stop bind bspace:bspace bind a:a bind numenter:numenter bind up:up bind n:n bind e:e bind num7:num7 bind voldn:voldn bind r:r bind num9:num9 bind u:u bind num4:num4 unbind mr bind caps:caps bind dot:dot bind p:p unbind g6 bind scroll:scroll bind play:play bind f4:f4 bind prtscn:prtscn bind pgup:pgup bind o:o bind f5:f5 unbind lock bind numlock:numlock bind space:space bind m:m bind rctrl:rctrl bind k:k bind lctrl:lctrl bind zone1:zone1 unbind g3 bind colon:colon bind lshift:lshift bind lwin:lwin bind g:g bind 6:6 bind h:h bind f6:f6 bind mute:mute bind del:del bind prev:prev bind num5:num5\nmacro clear\n\n", 1705) = 1705</code>
</details>

I already logged the value of `newlight->r[8] || newlight->g[8] || newlight->b[8]` before creating this issue. They're all set to `0`. That is what made me think of my patch actually.
```diff
if (kb->product == P_K55 || kb->product == P_K66 || kb->product == P_K68_NRGB) {
    // The K55 and K68 NRGB don't support winlock setting through the
    // normal packets, so we have to use a different packet to set it.
    // 8 is the winlock ("lock") led position in keymap.c
+   ckb_info("newlight->{r,g,b}: %d %d %d", newlight->r[8], newlight->g[8], newlight->b[8]);
    uchar winlock_status = (newlight->r[8] || newlight->g[8] || newlight->b[8]);
```
`journalctl -f` output:
```
ckb-next-daemon[7302]: [I] newlight->{r,g,b}: 0 0 0
```
This is **only** logged when the RGB updates, **not** when I press the <kbd>winlock</kbd> key. Disabling the RGB equals no more logs of the values.

The K55 KeyMap is based on the K95, its KeyMap does include a "lock" key:
https://github.com/ckb-next/ckb-next/blob/efbbae257749a22936af568e34ff9c527b7d0343/src/gui/keymap.cpp#L23-L24
and I can see it in the `Binding` tab:
[image]
Adding a `lock` key to the map did not change anything (don't know if I did that correctly I took the code from the K70MK2 KeyMap):
```diff
case KeyMap::K55:{
    // The K55 map is based on the K95
    map = getMap(KeyMap::K95, layout);

+   map["lock"].x = 58;
+   map["lock"].width = 10;
+   map["lock"].height = 8;
```
After removing the preceding code, I achieved to log the key properties with `qDebug` and got this output: `ckb-next[18471]: ckb-next: map of 'lock' x: 234, y: 0, width: 12, height: 12, hasLed: 1, hasScan: 1`. So I think the K55 KeyMap does have the `lock` light.

Ner0ox, 2023-03-04:
Skill issue

// thread closed

JotJoy, 2023-03-04:
Holy fucking bingle

tatokis, 2023-03-12:
Yes, apologies, you are correct in regards to the lock light already being there. Your GUI settings look correct too.

That said, looking at your strace logs, this is definitely a GUI bug. You should be seeing "rgb" commands.

Here's what mine looks like:
<details>
<summary>Enabling:</summary>
<code>[pid 88804] write(36, "rgb 0:ffffff 1:ffffff 2:ffffff 3:ffffff 4:ffffff 5:ffffff 6:ffffff 7:ffffff 8:ffffff 9:ffffff a:ffffff b:ffffff bslash:ffffff bspace:ffffff c:ffffff caps:ffffff colon:ffffff comma:ffffff d:ffffff del:ffffff dot:ffffff down:ffffff e:ffffff end:ffffff enter:ffffff equal:ffffff esc:ffffff f:ffffff f1:ffffff f10:ffffff f11:ffffff f12:ffffff f2:ffffff f3:ffffff f4:ffffff f5:ffffff f6:ffffff f7:ffffff f8:ffffff f9:ffffff g:ffffff grave:ffffff h:ffffff home:ffffff i:ffffff ins:ffffff j:ffffff k:ffffff l:ffffff lalt:ffffff lbrace:ffffff lctrl:ffffff left:ffffff light:ffffff lock:ffffff lshift:ffffff lwin:ffffff m:ffffff minus:ffffff mute:000000 n:ffffff next:ffffff num0:ffffff num1:ffffff num2:ffffff num3:ffffff num4:ffffff num5:ffffff num6:ffffff num7:ffffff num8:ffffff num9:ffffff numdot:ffffff numenter:ffffff numlock:ffffff numminus:ffffff numplus:ffffff numslash:ffffff numstar:ffffff o:ffffff p:ffffff pause:ffffff pgdn:ffffff pgup:ffffff play:ffffff prev:ffffff prtscn:ffffff q:ffffff quote:ffffff r:ffffff ralt:ffffff rbrace:ffffff rctrl:ffffff right:ffffff rmenu:ffffff rshift:ffffff rwin:ffffff s:ffffff scroll:ffffff slash:ffffff space:ffffff stop:ffffff t:ffffff tab:ffffff u:ffffff up:ffffff v:ffffff w:ffffff x:ffffff y:ffffff z:ffffff\n\n@1 rebind all notify all bind space:space bind rshift:rshift bind f1:f1 bind f9:f9 bind quote:quote bind z:z bind i:i bind pause:pause bind lwin:lwin bind num1:num1 bind m:m bind numslash:numslash bind f6:f6 bind f7:f7 bind num0:num0 bind prtscn:prtscn bind numlock:numlock bind 8:8 bind end:end bind v:v bind voldn:voldn bind f:f bind 9:9 bind rbrace:rbrace bind f3:f3 bind numenter:numenter bind minus:minus bind 0:0 bind rwin:rwin bind f11:f11 bind right:right bind scroll:scroll bind play:play bind g:g bind f10:f10 bind lctrl:lctrl bind equal:equal bind numstar:numstar bind s:s bind d:d bind enter:enter bind numminus:numminus bind num3:num3 bind k:k bind next:next bind rmenu:rmenu bind f8:f8 bind tab:tab bind x:x bind num7:num7 bind comma:comma bind f2:f2 bind stop:stop bind num9:num9 bind num5:num5 bind 4:4 bind grave:grave bind num4:num4 bind t:t bind num8:num8 bind b:b bind up:up bind 2:2 bind o:o bind e:e bind colon:colon bind home:home bind bspace:bspace bind ins:ins bind 1:1 bind f4:f4 bind u:u bind caps:caps bind y:y bind num2:num2 bind w:w bind p:p bind del:del bind pgup:pgup bind num6:num6 bind numdot:numdot bind left:left bind down:down bind dot:dot bind f12:f12 bind rctrl:rctrl bind esc:esc bind l:l bind bslash:bslash bind volup:volup bind pgdn:pgdn bind 5:5 bind 6:6 bind mute:mute bind 7:7 unbind light bind 3:3 bind a:a bind numplus:numplus bind ralt:ralt bind c:c bind slash:slash bind lbrace:lbrace bind lalt:lalt unbind lock bind lshift:lshift bind n:n bind f5:f5 bind h:h bind j:j bind prev:prev bind r:r bind q:q unbind lwin rwin\nmacro clear\n\n", 2835) = 2835</code>
</details>
<details>
<summary>Disabling:</summary>
<code>[pid 88804] write(36, "rgb 0:ffffff 1:ffffff 2:ffffff 3:ffffff 4:ffffff 5:ffffff 6:ffffff 7:ffffff 8:ffffff 9:ffffff a:ffffff b:ffffff bslash:ffffff bspace:ffffff c:ffffff caps:ffffff colon:ffffff comma:ffffff d:ffffff del:ffffff dot:ffffff down:ffffff e:ffffff end:ffffff enter:ffffff equal:ffffff esc:ffffff f:ffffff f1:ffffff f10:ffffff f11:ffffff f12:ffffff f2:ffffff f3:ffffff f4:ffffff f5:ffffff f6:ffffff f7:ffffff f8:ffffff f9:ffffff g:ffffff grave:ffffff h:ffffff home:ffffff i:ffffff ins:ffffff j:ffffff k:ffffff l:ffffff lalt:ffffff lbrace:ffffff lctrl:ffffff left:ffffff light:ffffff lock:000000 lshift:ffffff lwin:ffffff m:ffffff minus:ffffff mute:000000 n:ffffff next:ffffff num0:ffffff num1:ffffff num2:ffffff num3:ffffff num4:ffffff num5:ffffff num6:ffffff num7:ffffff num8:ffffff num9:ffffff numdot:ffffff numenter:ffffff numlock:ffffff numminus:ffffff numplus:ffffff numslash:ffffff numstar:ffffff o:ffffff p:ffffff pause:ffffff pgdn:ffffff pgup:ffffff play:ffffff prev:ffffff prtscn:ffffff q:ffffff quote:ffffff r:ffffff ralt:ffffff rbrace:ffffff rctrl:ffffff right:ffffff rmenu:ffffff rshift:ffffff rwin:ffffff s:ffffff scroll:ffffff slash:ffffff space:ffffff stop:ffffff t:ffffff tab:ffffff u:ffffff up:ffffff v:ffffff w:ffffff x:ffffff y:ffffff z:ffffff\n\n@1 rebind all notify all bind space:space bind rshift:rshift bind f1:f1 bind f9:f9 bind quote:quote bind z:z bind i:i bind pause:pause bind lwin:lwin bind num1:num1 bind m:m bind numslash:numslash bind f6:f6 bind f7:f7 bind num0:num0 bind prtscn:prtscn bind numlock:numlock bind 8:8 bind end:end bind v:v bind voldn:voldn bind f:f bind 9:9 bind rbrace:rbrace bind f3:f3 bind numenter:numenter bind minus:minus bind 0:0 bind rwin:rwin bind f11:f11 bind right:right bind scroll:scroll bind play:play bind g:g bind f10:f10 bind lctrl:lctrl bind equal:equal bind numstar:numstar bind s:s bind d:d bind enter:enter bind numminus:numminus bind num3:num3 bind k:k bind next:next bind rmenu:rmenu bind f8:f8 bind tab:tab bind x:x bind num7:num7 bind comma:comma bind f2:f2 bind stop:stop bind num9:num9 bind num5:num5 bind 4:4 bind grave:grave bind num4:num4 bind t:t bind num8:num8 bind b:b bind up:up bind 2:2 bind o:o bind e:e bind colon:colon bind home:home bind bspace:bspace bind [... cut ...]  c:c bind slash:slash bind lbrace:lbrace bind lalt:lalt unbind lock bind lshift:lshift bind n:n bind f5:f5 bind h:h bind j:j bind prev:prev bind r:r bind q:q\nmacro clear\n\n", 2818) = 2818</code>
</details>

You can see lock changes to ffffff and back to 000000.

With your K55 still in that if block, and the GUI running, can you see if these control the led?

```
echo "rgb lock:ffffff" > /dev/input/ckb1/cmd
```
```
echo "rgb lock:000000" > /dev/input/ckb1/cmd
```
assuming your keyboard is ckb1.

LockBlock-dev, 2023-03-17:
Checking for the correct ckb device:
```
$ cat /dev/input/ckb1/model
Corsair Gaming K55 RGB Keyboard
```
and trying to activate the lock LED:
```
$ echo "rgb lock:ffffff" > /dev/input/ckb1/cmd
```
And yes, this activates the LED like expected!
Disabling with `echo "rgb lock:000000" > /dev/input/ckb1/cmd` works too.

It must be a GUI issue like you said then. Do you have any idea what could be causing the problem? I'm open to search on my end but have no idea what exactly to look for.

tatokis, 2023-03-18:
The problem seems to be with
```cpp
        // Shift all keys to the left, and disable their LEDs
        QMutableHashIterator<QString, Key> i(map);
        while(i.hasNext()){
            i.next();
            i.value().x -= K95P_X_START;
            i.value().hasLed = false;
        }
```

The lock led is disabled, so the GUI doesn't send any rgb data for it.
Ideally, we need to have special handling for devices that don't have rgb lock lights, which would hide them in the lighting tab, but still allow on/off type control.

For now, right below that block above in the K55 keymap in keymap.cpp, add 
```cpp
map["lock"].hasLed = true;
```

LockBlock-dev, 2023-03-18:
Thanks! It works as expected. I replaced my messy fix with this one [in my fork](https://github.com/LockBlock-dev/ckb-next/commit/706f083f01145a360995891caf28cc8c02538f0f). The only issue is disabling the RGB disables the LED for the lock key too.

tatokis, 2023-03-29:
Feel free to make a commit with just these two changes and open a PR. The GUI definitely needs more work in this area, but that can be done later. It's better to have a working winlock led than not.