ckb-next/ckb-next #933: Device not found, Corsair K100 OPX optical switches, Please read before remove :)

Niklasw99, 2023-02-22:
### What kind of device is it?

Keyboard

### What is the name of the device?

K100 CORSAIR OPX

### Link to Corsair's Product Page

https://www.corsair.com/us/en/Categories/Products/Gaming-Keyboards/RGB-Mechanical-Gaming-Keyboards/K100/p/CH-912A01A-NA

### Additional Details

Add any additional details here.
i looked at the Request and there is no mention of this i looked around, this device is not Detected by CKB-NEXT 
Im running version ckb-next-git-1:0.5.0.r69.gc7d04173-1-x86_6

### Generated Report

[ckb-next-report-1677078742.gz](https://github.com/ckb-next/ckb-next/files/10805280/ckb-next-report-1677078742.gz)

gravityfargo, 2023-03-27:
It looks like there was an issue opened in May '22 about this device, but the issue contained two devices and was closed when the first one was taken care of #839 . I'll update the wiki to reflect this new issue.

Grocel, 2023-04-18:
More details about the K100 device are on this issue: https://github.com/ckb-next/ckb-next/issues/651
It is from October 2020.

gravityfargo, 2023-04-19:
I'll mark this as a duplicate of #651
I try not to touch the pre-2022 issues since they considerably predate my work on this project. Since that issue is still being actively posted on, I'll consider it the primary.
Wiki will be updated to reflect this.

closed 2023-04-19 by gravityfargo