ckb-next/ckb-next #937: K70 MK.2 RGB 'Save to hardware' greyed out

Lowrida, 2023-03-04:
### Describe your situation in detail.

Before i installed ckb-next i had 3 profiles, now i have only one left and i can't store any newly made profiles to the keyboard. So every time i boot my system i get a rainbow wave and can't change it to something else or adjust brightness. Very annoying when in bios or not logged into linux!

How can i store my own profiles to the device?

K70 MK.2 RGB
Linux Mint 20.3 Cinnamon

### Additional Details

Add any additional details here.

### Generated Report (if applicable)

Drag your generated report here if it's applicable.

Lowrida, 2023-03-08:
Anyone who knows how to do this?

Lowrida, 2023-03-12:
I guess nobody knows...

Ravenslofty, 2023-03-12:
It's greyed out because we haven't implemented it in ckb-next yet.

Lowrida, 2023-03-12:
Ah ok. But why are my other profiles gone? Standard there were three of them..

gravityfargo, 2023-03-27:
I don't know enough about the code to answer why the other profiles are gone, but I'd like to comment we can't give a timeline for any particular feature. This project is volunteer-run, and those that contribute to the code base do so on their own time.

Lowrida, 2024-02-23:
Almost a year later i wonder if this is implemented?

gravityfargo, 2024-05-01:
From my previous comment:
> I'd like to comment we can't give a timeline for any particular feature. This project is volunteer-run, and those that contribute to the code base do so on their own time.

You did not include a generated report, so I can't tell which specific product ID you are referring to, but the three product IDs listed in the wiki for the K70's indicate:
> hardware playback not supported. Settings will be saved to software only.

All that said, you are welcome to contribute code and submit a pull request. I am closing this issue.

closed 2024-05-01 by gravityfargo