ckb-next/ckb-next #939: New Device Request: Corsair CORSAIR K55 RGB PRO Gaming Keyboard

keilmillerjr, 2023-03-18:
### What kind of device is it?

Keyboard

### What is the name of the device?

Corsair CORSAIR K55 RGB PRO Gaming Keyboard

### Additional Details

Thank you.

ckb-next-dev-detect - Sat Mar 18 07:27:39 EDT 2023

```
Kernel: Linux archlinux 6.2.6-arch1-1 #1 SMP PREEMPT_DYNAMIC Mon, 13 Mar 2023 17:02:08 +0000 x86_64 GNU/Linux

OS Version:
LSB Version:	n/a
Distributor ID:	Arch
Description:	Arch Linux
Release:	rolling
Codename:	n/a

USB Devices:

Bus 001 Device 007: ID 1b1c:0c1a Corsair CORSAIR Lighting Node CORE
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c1a 
  bcdDevice            0.09
  iManufacturer           1 Corsair
  iProduct                2 CORSAIR Lighting Node CORE
  iSerial                 3 1670020C22004BAF4FAC835CC01B00F5
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower               50mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      33
[... 203 lines cut ...]

Dev nodes:
/dev/input/ckb0
/dev/input/ckb0/pid
689
/dev/input/ckb0/version
0.5.0
/dev/input/ckb0/connected

Kernel command line:
initrd=\initramfs-linux.img root=PARTUUID=c4185cd7-52c6-4000-b523-ff3fd4de196b rw

GUI ldd:

Environment:
XDG_SESSION_DESKTOP=gnome
QT_QPA_PLATFORMTHEME=qt5ct
XDG_SESSION_TYPE=x11
XDG_CURRENT_DESKTOP=GNOME
DISPLAY=:1
```

### Generated Report

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/11008368/ckb-next-dev-detect-report.gz)

JanNowak94, 2023-03-19:
use git version (ckb-next-git package) - not stable (have also done same mistake).
If anything I will just note that lights are still broken (wrong zones and winlock is treated as normal zone) and remaping fn to right super (LOGO/Meta[kde]) key results in unusable modifier [but I could not pinpoint why] - but remapping Fn to menu and menu to right super does work, except part where you are pushing wrong button repeatedly ...

keilmillerjr, 2023-03-19:
@JanNowak94 Thank you. I am not sure why I didn't think to grab the git version. My keyboard is now recognized. I created a new "mode", changed all 5 zones to lime... and the right side of my keyboard is different. Nav is orange and num is red. Any ideas?

closed 2023-03-22 by keilmillerjr