ckb-next/ckb-next #940: CKN-Next-Daemon fails to start

ajgp, 2023-03-18:
### Describe your situation in detail.

The CKB-NEXT-DAEMON fails to start either on boot or when manually started, and application doesnt detect any corsair devices despite dmesg finding a supported mouse (Katar pro xt, product ID 0x1bac)

Output from systemctl status ckb-next-daemon is as follows

× ckb-next-daemon.service - Corsair Keyboards and Mice Daemon
     Loaded: loaded (/usr/lib/systemd/system/ckb-next-daemon.service; enabled; preset: disabled)
     Active: failed (Result: core-dump) since Sat 2023-03-18 12:14:41 GMT; 4min 13s ago
   Duration: 577ms
    Process: 1872 ExecStart=/usr/libexec/ckb-next-daemon (code=dumped, signal=ABRT)
   Main PID: 1872 (code=dumped, signal=ABRT)
        CPU: 47ms

Mar 18 12:14:41 DESKTOP-1M993LH.lan systemd[1]: ckb-next-daemon.service: Scheduled restart job, restart counter is at 5.
Mar 18 12:14:41 DESKTOP-1M993LH.lan systemd[1]: Stopped Corsair Keyboards and Mice Daemon.
Mar 18 12:14:41 DESKTOP-1M993LH.lan systemd[1]: ckb-next-daemon.service: Start request repeated too quickly.
Mar 18 12:14:41 DESKTOP-1M993LH.lan systemd[1]: ckb-next-daemon.service: Failed with result 'core-dump'.
Mar 18 12:14:41 DESKTOP-1M993LH.lan systemd[1]: Failed to start Corsair Keyboards and Mice Daemon.

### Additional Details

Add any additional details here.
CPU: Ryzen 5 3600
RAM: 32GB
GPU: AMD 6700XT
O/S: Opensuse Tumbleweed
uname -a output: : Linux DESKTOP-1M993LH.lan 6.2.4-1-default #1 SMP PREEMPT_DYNAMIC Sat Mar 11 10:13:47 UTC 2023

Corsair Product: Katar Pro XT,    DMESG idenifies it as product id 1bac  which should be supported.

### Generated Report (if applicable)

Drag your generated report here if it's applicable.
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/11008503/ckb-next-dev-detect-report.gz)

tatokis, 2023-03-18:
Please run `coredumpctl info ckb-next-daemon -1` and paste the output here.

ajgp, 2023-03-18:
output of that command:

           PID: 6589 (ckb-next-daemon)
           UID: 0 (root)
           GID: 0 (root)
        Signal: 6 (ABRT)
     Timestamp: Sat 2023-03-18 12:38:41 GMT (1min 20s ago)
  Command Line: /usr/libexec/ckb-next-daemon
    Executable: /usr/libexec/ckb-next-daemon
 Control Group: /system.slice/ckb-next-daemon.service
          Unit: ckb-next-daemon.service
         Slice: system.slice
       Boot ID: 21d67c70e598443aa8eaf909102fd2a0
    Machine ID: 473a5790d0194ed7b06b07df8f3a7fb4
      Hostname: DESKTOP-1M993LH.lan
       Storage: /var/lib/systemd/coredump/core.ckb-next-daemon.0.21d67c70e598443aa8eaf909102fd2a0.6589.1679143121000000.zst (present)
  Size on Disk: 49.4K
       Message: Process 6589 (ckb-next-daemon) of user 0 dumped core.
                
                Stack trace of thread 6591:
                #0  0x00007f5b053c047c __pthread_kill_implementation (libc.so.6 + 0x9047c)
                #1  0x00007f5b0536de16 raise (libc.so.6 + 0x3de16)
                #2  0x00007f5b0535689c abort (libc.so.6 + 0x2689c)
                #3  0x00007f5b053575d7 __libc_message.cold (libc.so.6 + 0x275d7)
                #4  0x00007f5b05455c4b __fortify_fail (libc.so.6 + 0x125c4b)
                #5  0x00007f5b054540c6 __chk_fail (libc.so.6 + 0x1240c6)
                #6  0x00007f5b05453c35 __snprintf_chk (libc.so.6 + 0x123c35)
                #7  0x0000559ff8e84003 n/a (ckb-next-daemon + 0xd003)
                #8  0x0000559ff8e84240 n/a (ckb-next-daemon + 0xd240)
                #9  0x0000559ff8e9e077 n/a (ckb-next-daemon + 0x27077)
                #10 0x00007f5b053be615 start_thread (libc.so.6 + 0x8e615)
                #11 0x00007f5b05444c60 __clone3 (libc.so.6 + 0x114c60)
                
                Stack trace of thread 6589:
                #0  0x00007f5b0543983c __select (libc.so.6 + 0x10983c)
                #1  0x0000559ff8e7fcdd n/a (ckb-next-daemon + 0x8cdd)
                #2  0x00007f5b05357af0 __libc_start_call_main (libc.so.6 + 0x27af0)
                #3  0x00007f5b05357bb9 __libc_start_main@@GLIBC_2.34 (libc.so.6 + 0x27bb9)
                #4  0x0000559ff8e804f5 n/a (ckb-next-daemon + 0x94f5)
                
                Stack trace of thread 6592:
                #0  0x00007f5b05438abf ioctl (libc.so.6 + 0x108abf)
                #1  0x0000559ff8e9986d n/a (ckb-next-daemon + 0x2286d)
                #2  0x00007f5b053be615 start_thread (libc.so.6 + 0x8e615)
                #3  0x00007f5b05444c60 __clone3 (libc.so.6 + 0x114c60)
                
                Stack trace of thread 6590:
                #0  0x00007f5b05407ce5 clock_nanosleep@GLIBC_2.2.5 (libc.so.6 + 0xd7ce5)
                #1  0x0000559ff8e93eae n/a (ckb-next-daemon + 0x1ceae)
                #2  0x00007f5b053be615 start_thread (libc.so.6 + 0x8e615)
                #3  0x00007f5b05444c60 __clone3 (libc.so.6 + 0x114c60)
                ELF object binary architecture: AMD x86-64

closed 2023-03-18 by tatokis

tatokis, 2023-03-18:
I'll ask OpenSUSE to backport the fix. Hopefully you'll receive an update in a few days that fixes it.

commit c29a9f5e (2023-03-18): Fix buffer overflow detected with _FORTIFY_SOURCE

Technically there's no buffer overflow taking place, but the size
argument passed to snprintf was incorrect.

Closes #940