ckb-next/ckb-next #943: wiki/Supported-Hardware add minimal version of driver with support.

JanNowak94, 2023-03-21:
### Enhancement Details

As in title - this should lower frequency of questions .. Probably ...

### Additional Details

Add any additional details here.

JanNowak94, 2023-03-23:
I needed patebin it (github "something went really wrong").
https://pastebin.com/HqHkg8cf
There is issue with Strafe Strafe MK.2 - it seems to use same device id as RGB - so I cannot check when it was added.
Preview:

\* = hardware playback not supported. Settings will be saved to software only.

If your device isn't on this list, it isn't currently supported by ckb-next. Pull requests welcome.

If your device is listed as *Experimental*, this means that support for it is not enabled by default, and is  incomplete, or possibly even broken. If you wish to proceed, you can read [here](https://github.com/ckb-next/ckb-next/wiki/Linux-Installation#enabling-support-for-experimental-devices) on how to enable support for it.

Please open a bug report if you find one of these variants incompatible or an Enhancement issue if your variants is compatible.
## Keyboards

| FAMILY   | MODEL                      | Status            | DEVICE ID   | From version |
|----------|----------------------------|-------------------|-------------|--------------|
|**K55**   | RGB                        | *Supported        | 1b3d        | 3.0          |
|          | RGB PRO                    | *Supported        | 1ba4        | git          |
|          | RGB PRO LITE               |                   |             |              |
|          | RGB PRO XT                 | *Supported        | 1ba1        | git          |
|**K57**   | Wireless                   | *Experimental     |             |              |
|**K60**   | RGB PRO Low Profile        | *Supported        | 1bad        | 5.0          |
|          | RGB PRO SE                 | *Supported        | 1b8d        | 5.0          |
|          | RGB PRO                    | *Supported        | 1ba0        | 5.0          |
|          | PRO (Monochrome)           | *Supported        | 1b83        | git          |
|          | VENGEANCE                  |                   |             |              |
|**K63**   | Compact                    | Supported         | 1b40        | 3.0          |
|          | Wireless                   | *Experimental     |             | 5.0          |
|          | Wireless Special Edition   | *Experimental     |             | 5.0          |
|**K65**   | Mini                       |                   |             |              |
|          | LUX                        | Supported         | 1b37        | 3.0          |
|          | RapidFire                  | Supported         | 1b39        | 3.0          |
|          | Compact                    |                   |             |              |
|          | VENGEANCE                  |                   |             |              |
|**K66**   | RGB                        | Supported         | 1b41        | 3.2          |
|**K68**   | RGB                        | Supported         | 1b4f        | 3.0          |
|          | RAPIDFIRE                  | Supported         | 1b3f        | 3.0          |
|**K70**   | RGB                        | Supported         | 1b13        | 3.0          |
|          | TKL CHAMPION SERIES        | Requested #745    | 1b73        |              |
|          | PRO                        |                   |             |              |
|          | MK.2 Low Profile           | *Supported        | 1b55        | 4.0          |
|          | MK.2 Low Profile RAPIDFIRE | *Supported        | 1b55        | 4.0          |
|          | MK.2                       | *Supported        | 1b49        | 3.1          |
|          | MK.2 SE                    | *Supported        | 1b6b        | 3.1          |
|          | LUX                        | Supported         | 1b36        | 3.0          |
|          | LUX RGB                    | Supported         | 1b33        | 3.0          |
|          | RGB RAPIDFIRE              | Supported         | 1b38        | 3.0          |
|          | RAPIDFIRE                  | Supported         | 1b3a        | 3.0          |
|**K90**   | VENGEANCE                  | Supported         | 1b02        | 3.0          |
|**K95**   | PLATINUM                   | *Supported        | 1b2d        | 3.0          |
|          | PLATINUM SE                |                   |             |              |
|          | PLATINUM XT                | *Supported        | 1b89        | 5.0          |
|          | RGB                        | Supported         | 1b11        | 3.0          |
|**K100**  | RGB Optical-Mechanical     | Requested #839    | 1b7c        |              |
|          | RGB Mechanical             |                   | 1b7d        |              |
|**STRAFE**| NonRGB1                    | Supported         | 1b15        | 3.0          |
|          | NonRGB2                    | Supported         | 1b44        | 3.0          |
|          | RGB                        | Supported         | 1b20        | 3.0          |
|          | MK.2                       | Supported         |             | ?            |

## Mice
| Family       | MODEL                   | STATUS              | Device ID    | From version |
|--------------|-------------------------|---------------------|--------------|--------------|
| M55          | PRO                     | *Supported          | 1b70         | 5.0          |
| M65          | RGB                     | Supported           | 1b12         | 3.0          |
|              | PRO                     | Supported           | 1b2e         | 3.0          |
[... cut ...]
|VIRTUOSO |RGB WIRELESS SE|Headset          | Requested #785              |             |              |
|VOID     |PRO RGB        |Headset          | Requested #735              | 0a14        |              |
|         |Slipstream     |Wireless Reciever| Requested #785              | 0a3e        |              |
|         |Commander Core |Controller Hub   | Requested #833              | 0c1c        |              |
|         |Commander PRO  |Controller Hub   | Requested #689              | 0c10        |              |

gravityfargo, 2023-03-28:
I've updated the wiki since your last post. I push my changes to [a gist of mine](https://gist.github.com/gravityfargo/2fbce071791a4ea015b27a33a4abf534) before I make the changes on the wiki proper. 

I've been mulling over some way to make 
> * = hardware playback not supported. Settings will be saved to software only.

more obvious, and

> "Please note that version listed below is the absolute minimum required. It is not the recommended version for your device.".

would be in that same mindset. 

When I have the will for it I'll make some pretty graphics and whatnot. Anyway, I'll work on these changes when @tatokis or another person more qualified than me gives the thumbs up.

tatokis, 2023-03-29:
@JanNowak94 Thank you very much for this. Really appreciated.
The Strafe RGB MK.2 is `0x1b48`. No device shares a usb id with another one.

@gravityfargo
Maybe change Status to STATUS as well.

A few nitpicks:
1. I think `From version` would sound better as either `Since`, `Since version`, or if not too long, `Supported since`.
2. Instead of `git`, maybe put `0.5.1 (git)`.

Other than that, I fully agree with:

> At worst this documentation is still better - it does tell user absolute minimum that he should expect anything from.

So go ahead and make any changes you deem appropriate. If something is incorrect, we'll fix it later down the line.

> "Please note that version listed below is the absolute minimum required. It is not the recommended version for your device.".

Maybe if possible also clarify that this can be dependent on the firmware version on the device itself. Usually though, once a device is added to a release, it functions adequately. Otherwise it's stuck in experimental hell.

gravityfargo, 2023-04-21:
I have made changes have been made to the wiki

closed 2023-04-21 by gravityfargo