ckb-next/ckb-next #944: Wayland Support

AFCMS, 2023-03-29:
### Enhancement Details

Currently, windows must use Xwayland in order to be detected and allow switching profile.

It is quite annoying, because more and more distros use Wayland by default, including Ubuntu, and many apps start supporting it natively.

Blender enabled Wayland support by default, so to get my profile to switch I needed to manually disable Wayland support.

Some apps even have Wayland as a requirement now, like Waydroid.

It would be nice to get full Wayland support into CKB-Next.

### Additional Details

https://code.blender.org/2022/10/wayland-support-on-linux/
https://waydro.id/

tatokis, 2023-03-29:
Without going into a rant; last I checked, in typical Wayland fashion, the people behind it consider this a security issue and a niche, and they seem to not care at all. I'd be surprised if this changed at all, but I'd love to be proven wrong.

I heard sway made their own protocol to support this use case, but I never looked into it.

In short:
* If you use sway: Patches welcome.
* If you use anything else without this functionality (like GNOME or KDE): Complain to them (and if you're told that they have implemented this in a unified/cross DE way, let me know).

Any pure Wayland support in ckb-next (outside of XWayland) is purely accidental due to Qt and should be considered untested/unsupported. Just like Wayland doesn't care about "niches", I don't care to support it.

tl;dr use X.

closed 2023-03-29 by tatokis

AFCMS, 2024-05-16:
If someone comes around this issue, please upvote https://github.com/flatpak/xdg-desktop-portal/issues/304