ckb-next/ckb-next #947: Add support for K60 Pro TKL

ItsCryptic, 2023-03-31:
### What kind of device is it?

Keyboard

### What is the name of the device?

K60 Pro TKL RGB Optical-Mechanical Gaming

### Link to Corsair's Product Page

https://www.corsair.com/us/en/Categories/Products/Gaming-Keyboards/RGB-Mechanical-Gaming-Keyboards/K60-PRO-TKL-RGB-Gaming-Keyboard/p/CH-911D01A-NA

### Additional Details

Add any additional details here.

### Generated Report

Drag your generated report here.
[ckb-next-report-1680292007.zip](https://github.com/ckb-next/ckb-next/files/11125431/ckb-next-report-1680292007.zip)

Cornerstar31, 2023-04-01:
I want to put my K60 in white only mode without a GUI in proxmox.
Is it possible to do it with a few bash commands.
I truly hate all those flashing colors

gravityfargo, 2023-04-05:
> I want to put my K60 in white only mode without a GUI in proxmox. Is it possible to do it with a few bash commands. I truly hate all those flashing colors

As far as I know, no. I'm not a dev but the documentation says QT frameworks are required to build. I think your only option is to fire up a windows installation and change the hardware profile in iCUE. Don't quote me but I think there is a USB passthrough function for proxmox VMs. Proxmox is just a variation of Debian anyway. 

In the future, please open a separate issue

gravityfargo, 2025-05-13:
Support added in v0.6.0 e8864f8

closed 2025-05-13 by gravityfargo