ckb-next/ckb-next #950: Corsair M65 Ultra Wireless Support

rastla, 2023-04-08:
### What kind of device is it?

Mouse

### What is the name of the device?

Corsair M65 RGB Ultra Wireless

### Link to Corsair's Product Page

https://www.corsair.com/us/en/Categories/Products/Gaming-Mice/FPS-Fast-Action-Mice/M65-RGB-ULTRA-WIRELESS-Tunable-FPS-Gaming-Mouse/p/CH-9319411-NA2

### Additional Details

Add any additional details here.

### Generated Report

Drag your generated report here.
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/11182183/ckb-next-dev-detect-report.gz)

gravityfargo, 2023-04-08:
Could you unplug your slipstream dongle, plug the mouse in via USB, then rerun the report? Slipstream hijacks names in the report. Thanks!

IcyStarsy, 2023-04-14:
[ckb-next-report-1681479587.gz](https://github.com/ckb-next/ckb-next/files/11233457/ckb-next-report-1681479587.gz)

padrino121, 2023-09-18:
Attaching another as well in case it helps
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/12644526/ckb-next-dev-detect-report.gz)

MysteriousRSA, 2024-01-14:
I'd like to know if there's been any update on this

N0madical, 2024-01-29:
I would like to second this, I'll provide some files if it's helpful:

Debug with mouse plugged in via USB:
[ckb-next-report-1706489558.gz](https://github.com/ckb-next/ckb-next/files/14078540/ckb-next-report-1706489558.gz)

Debug with mouse connected via Bluetooth:
[ckb-next-report-1706489418.gz](https://github.com/ckb-next/ckb-next/files/14078543/ckb-next-report-1706489418.gz)

Thank you for making ckb-next!

AFCMS, 2024-02-22:
I have this device and a bit of technical knowledge with C++.

I might try implementing support, but I don't know where to start.

If someone try to implement support I will be able to test too.

N0madical, 2024-06-18:
Would it be possible for this device to have support via the cable, just not slipstream?

gravityfargo, 2024-07-12:
If someone were to submit a PR adding support for this mouse, it will most definitely be for USB only. Slipstream requires a lot of back end work to support any devices at all.