ckb-next/ckb-next #953: Random Play/Pause/Volume change

mysteryx93, 2023-04-11:
### Affected Devices

K95 RGB Platinum XT

### Linux Distribution

Garuda (arch-based)

### Installation Method

chaotic-aur

### ckb-next version

ckb-next 0.5.0

### What did you expect?

I'm having occasional issues of random play/pause while watching videos, of volume swinging up or down. It happens occasionally, could go the whole without issue and then the audio goes berserk. Only happens when watching videos in YouTube or MPlayer. I have my own audio player software that is not affected by the issue (it doesn't listen to keyboard media keys).

Since stopping service `ckb-next-daemon`, I haven't had the issue, so I believe that `ckb-next` is responsible.

The problem still occurs without the Corsair keyboard connected. With a different keyboard, the problem continues.

### Did you try anything that may have helped?

I tried:
- Changing kernel
- Closing "competing" audio apps
- Switching keyboard
- Closing `ckb-next` app (with service still running)

Stopping `ckb-next-daemon` seems to be the solution.

### Additional Details

Add any additional details here.

### Generated Report

Drag your generated report here.
[ckb-next-report-1681242266.gz](https://github.com/ckb-next/ckb-next/files/11204338/ckb-next-report-1681242266.gz)

tatokis, 2023-04-14:
> The problem still occurs without the Corsair keyboard connected. With a different keyboard, the problem continues.

> Stopping ckb-next-daemon seems to be the solution.

This doesn't make sense to me. Unless ckb-next-daemon is hung somewhere, when the device is disconnected it can't send any input events because it completely removes the virtual input devices.

Try running ckb-next-daemon with no corsair devices attached, and when the bug happens, run and share the output of `xinput`.

tatokis, 2023-04-14:
It'd also be useful if you kept `evtest` running against the emulated devices, and see if any `KEY_VOLUMEUP`/`KEY_VOLUMEDOWN` are generated by the daemon when this happens.

mysteryx93, 2023-04-27:
After disabling ckb-next keyboard service, the problem occurred a lot less; but it's still happening, it seems more with Vivaldi web browser.

So it seems that the problem is triggered by something that is used by `ckb-next` and `vivaldi`. But the problem is neither of these.

Any idea what this could be??