ckb-next/ckb-next #959: Katar Pro XT (wired version) not working on Fedora 38

MaxTheBearMan, 2023-04-29:
### Describe your situation in detail.

Hello I'm running Fedora 38 with a katar pro xt mouse. When installing ckb-next via dnf my mouse immediately stops working until I either turn off the daemon with systemctl, or I switch usb ports. However, no matter what I do the device will not be detected by the ckb-next application, even after a reboot. 

Very strangely if I instead choose to install ckb-next via the quickinstall script on github, the device will show up as detected, but otherwise remains in an unusable state. All the physical buttons on the mouse work, but when trying to move the mouse the cursor will just dart to the right side of my displays and stay there. Extremely bizarrely while the cursor won't respond to moving the mouse, I can get it to move left and right by scrolling up and down on the scroll wheel, but as soon as I move the mouse at all the cursor just darts back to the right of the screens. 

To be clear regardless of what version of the program I use the mouse does work fine as long as the ckb-next daemon has been stopped, or as long as ckb-next is completely uninstalled. I just want to be able to turn off the dang rgb light and change my dpi settings, neither of which can I do without ckb-next

Interestingly just last week I was on Manjaro using the same mouse with the aur git version of ckb-next and everything was working fine. Other than being on Fedora now the only other thing I can think of that could have changed is that Fedora 38 had me install new firmware for the mouse upon my first update. The mouse is on version 1.6.26 now from lvfs I think. I'm not sure what it was on before. I wish I could revert it to an older firmware to see if that fixes things, but I haven't the slightest idea how.

Hopefully this is the correct place for this. I've never really posted a bug report before, and figured I should use the help request feature before using the bug report feature as I'm not even sure whether this is as bug with ckb-next or if it somehow has something to do with Fedora, or the new mouse driver.

Thank you in advance for any assistance

### Additional Details

Add any additional details here.

### Generated Report (if applicable)

Drag your generated report here if it's applicable.
[ckb-next-report-1682783842.gz](https://github.com/ckb-next/ckb-next/files/11359580/ckb-next-report-1682783842.gz)

tatokis, 2023-04-30:
The new descriptors
```
001:013:003:DESCRIPTOR         1682811900.877088
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 19 00 29 97 15 00 25 01 75 01
 95 98 81 02 05 08 19 01 29 03 25 01 75 01 95 03
 91 02 75 05 95 01 91 01 C0

001:013:002:DESCRIPTOR         1682811900.880109
 06 42 FF 09 02 A1 01 15 00 26 FF 00 75 08 95 40
 09 02 81 02 C0

001:013:001:DESCRIPTOR         1682811900.883107
 06 42 FF 09 01 A1 01 15 00 26 FF 00 75 08 95 40
 09 01 81 02 95 40 09 01 91 02 C0

001:013:000:DESCRIPTOR         1682811900.887097
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 95 05 75 01 81 02 95 03 75 01 81 03
 75 10 95 02 05 01 09 30 09 31 16 01 80 26 FF 7F
 81 06 75 08 95 01 05 01 09 38 15 81 25 7F 81 06
 06 00 FF 09 F1 75 08 95 05 15 00 26 FF 00 81 00
 C0 C0
```

aussyspitz, 2023-04-30:
I have the exact same mouse everything works except the dpi button won't loop with ckb-next. I  dual boot with Windows I downloaded the latest i-cue software from Corsair and the latest firmware is showing 1.5.24.  Before the mouse arrived I saw a ton of complaints regarding firmware version 1.6.26 and people wanting to force rollback to version 1.5.24. Not 100% but I'm pretty sure you may have to revert/force using a windows machine I even found a youtube video that had a way to manually load the new firmware to force a rollback on Windows but since I didn't need to do that on mine I didn't save the link. Be advised there are at least two different versions of this mouse looks like one for EU/NA and the other for Asia it was unclear if they use different hardware or require different firmware. I will try and see if I can relocate the rollback discussion and the youtube video for manual rollback.

aussyspitz, 2023-04-30:
Looks like Corsair pulled firmware 1.6.26 due to so many complaints but didn't really offer a solution.  There were several solutions offered in the Corsair forum https://forum.corsair.com/forums/topic/181349-corsair-katar-pro-xt-wired-mouse-possible-issue-with-new-1626-firmware/   You could also contact Corsair as they might be able to send you a bin file of 1.5.24 to update via windows the youtube video I found was not in English so you would kind of have to wing it but it looks like it helped some people he has a direct link to an older firmware version 1.2.23   https://www.youtube.com/watch?v=g9RG4gY9O1I

aussyspitz, 2023-05-01:
Link to comment of someone who fixed it https://forum.corsair.com/forums/topic/181349-corsair-katar-pro-xt-wired-mouse-possible-issue-with-new-1626-firmware/?do=findComment&comment=1035656 Here is Link to referenced icue version 3.38.88 I downloaded and scanned with Webroot AV came up clean it is a Windows.msi not sure if you can run it under wine or if you would have to use a windows machine but it should get you back to v1.5.24 firmware https://www.touslesdrivers.com/php/constructeurs/redirection.php?v_code=70896

MaxTheBearMan, 2023-05-02:
woah. Thanks. I'll definitely check out those links later in the week, I'll have to get permission to use someones windows computer, one of my brothers may oblige me. Man, it's real annoying Fedora had me download the latest firmware given it's problematic.

OSCrustacean, 2023-05-06:
I have the same mouse, had to do a forced firmware rollback using windows on a separate device and am experiencing the same issue. I also have this issue with the K70 RGB Rapidfire keyboard, it stops working immediately after installing ckb-next on Fedora 38 (and neither are recognized). Everything still works on Fedora 37 and also on Ubuntu 23.04 that I have tested.

tatokis, 2023-05-06:
> it stops working immediately after installing ckb-next on Fedora 38 (and neither are recognized).

https://bugzilla.redhat.com/show_bug.cgi?id=2192159

aussyspitz, 2023-05-06:
> I have the same mouse, had to do a forced firmware rollback using windows on a separate device and am experiencing the same issue. I also have this issue with the K70 RGB Rapidfire keyboard, it [...]

Thanks for chiming in OSCrustacean. I believe based on your experience that the OP should still roll their firmware back to v1.5.24 via windows correct?

OSCrustacean, 2023-05-11:
> Thanks for chiming in OSCrustacean. I believe based on your experience that the OP should still roll their firmware back to v1.5.24 via windows correct even if it doesn't fix the issue for Fedora 38?

Yes, I would agree with that because it seemed to have issues on other distros with that firmware.

OSCrustacean, 2023-05-12:
Not sure exactly what happened, but I've checked in periodically and my setup works completely fine as of a fresh install today

aussyspitz, 2023-05-13:
> > I'm curious my mouse works but my DPI button won't loop it will cycle fine from 400 to 3200 my highest DPI setting but then it won't cycle back/through to 400. It works perfectly fine and as expected on windows so its not a button issue but it doesn't work on linux. Does yours do the same or does it fully cycle/loop?
> 
> Yeah now that you mention it mine does the same actually, I hadn't tried that yet.
> 
> EDIT: Interestingly enough in the app you can set the button to cycle down instead of up, and that seems to work fine. EDIT 2: I think I see what's happening, the button seems to default to up instead of cycle up, because when I set it to cycle up it worked as expected.

You are a life saver that fixed my issue works perfectly thank you x1000

MaxTheBearMan, 2023-05-15:
> Link to comment of someone who fixed it https://forum.corsair.com/forums/topic/181349-corsair-katar-pro-xt-wired-mouse-possible-issue-with-new-1626-firmware/?do=findComment&comment=1035656 Here is download link to referenced icue version 3.38.88 I downloaded and scanned with Webroot AV came up clean it is a Windows.msi not sure if you can run it under wine or if you would have to use a windows machine but it should get you back to v1.5.24 firmware https://www.touslesdrivers.com/php/constructeurs/redirection.php?v_code=70896

I know it's been a while for me to report back, but I can now confirm that rolling back the firmware via the older version of ICUE via a family members windows PC has solved the problem. My advice to any other Fedora users who come upon this issue is for them to uncheck lvfs as a source from their software store and then take your mouse to a windows pc, download the older version of ICUE (version 3.38.88) as mentioned above (though I got it direct from the Fedora website rather than the above link), rollback the firmware, and then plug the mouse back into their Fedora machine and all should be good to go.

aussyspitz, 2023-05-16:
Glad to hear you got it fixed and appreciate you coming on and letting everyone know I'm sure your info will be very helpful for others with the same issue

gravityfargo, 2024-05-02:
I added this as a known issue in the wiki. Thank you all for finding a solution and reporting back!

Kangaroux, 2024-10-26:
@gravityfargo FWIW I also experienced these same symptoms on Pop!_OS 22.04.

# How to fix
tl;dr for anyone coming here looking for a solution: Firmware v1.6.26 is buggy and downgrading to v1.5.24 fixes the issue.

I've uploaded a copy of the v1.5.24 firmware that you can flash directly onto your mouse. No tools needed, it's quick to do. If you don't feel comfortable doing that the longer version involves using an old version of iCUE (can be done in Virtualbox).

## Method 1: Manual flash
Get a paperclip or toothpick. Unplug the Katar Pro XT, insert said paperclip/toothpick into the small hole on the bottom of the mouse, and then plug the device back in. You should see a USB storage device named CRP DISABLD that's been mounted. **The mouse will not work when it's in this mode, that's normal.** Copy `firmware.bin` from the device to your machine as a backup. Download [firmware1.5.24.zip](https://github.com/user-attachments/files/17528407/firmware1.5.24.zip), extract, and move onto the device. Re-plugin your mouse and you should be good to go.

```bash
# v1.5.24
$ sha1sum firmware.bin
e4a35601d603ffb91b2afa76ab2ff5010ef50395  firmware.bin
```

## Method 2: Flash using legacy iCUE
iCUE v3.38.88 has the version of firmware we need. You can get it here: http://downloads.corsair.com/Files/CUE/iCUESetup_3.38.88_release.msi

It can be done in Virtualbox without too much hassle, which I've outlined below.

1. Add yourself to the `vboxusers` group so Virtualbox can see USB devices: `sudo usermod $USER -aG vboxusers`
2. Logout then log back in, run `groups`, and verify you see `vboxusers`. If you don't, do a full reboot (seems to be an issue on Ubuntu).
3. Install iCUE from the link above and launch it, it should say no device found
4. (Recommended) Plugging in a second mouse makes this next part easier. If you don't have one you can still do it by pressing tab/space/enter to navigate
5. At the top of your vbox window go to Devices > USB > CORSAIR KATAR PRO XT. If you don't see this you probably are missing the [Virtualbox extension pack](https://adamtheautomator.com/vbox-usb-passthrough/)
6. Your device should have appeared in iCUE
7. Click Settings, and check for firmware updates. It should say v1.5.24. Start the update. The RGB on the mouse should turn white and start blinking
8. If the mouse disappears before the update starts, repeat step 5, then start the update again
9. It should flash the new firmware and it will seem to get stuck at 100%. I waited about 10 seconds and then replugged in the mouse and it was fine. 
10. Close the VM, open ckb-next and verify the new firmware version. Your cursor also shouldn't get stuck now when ckb-next is running

ericherman, 2025-05-16:
Firmware v1.7.0 seems to have similar issues as v1.6 firmwares