ckb-next/ckb-next #960: No config unless application is explicitely launched

maplepy, 2023-04-30:
### Affected Devices

m65 k70

### Linux Distribution

Arch

### Installation Method

Aur

### ckb-next version

ckb-next 0.5.0-3

### What did you expect?

Keyboard and mouse macros/rebinds and config works 

If I do not start the application, nothing works. Although I have the daemon and the process launched on boot with `systemctl enable`

### Did you try anything that may have helped?

1. Tried launching the daemon with sudo
2. Reinstalling
3. Disabling the process on boot
4. Switching to the git version
5. Replug
6. Switch usb port

### Additional Details

Add any additional details here.

### Generated Report

[image]

poperigby, 2024-06-06:
Did you ever figure this out?

maplepy, 2024-06-11:
Well unfortunately no, the devs are inactive too so yeah :/