ckb-next/ckb-next #964: Add support for Corsair HS80 RGB Wireless

BorisBrock, 2023-05-14:
### What kind of device is it?

Headphone

### What is the name of the device?

Corsair HS80 RGB Wireless

### Link to Corsair's Product Page

https://www.corsair.com/de/de/Kategorien/Produkte/Gaming-Headsets/Kabellose-Headsets/HS80-RGB-WIRELESS-Premium-Gaming-Headset-with-Spatial-Audio/p/CA-9011235-EU

### Additional Details

The headset has two controllable LEDs:

- RGB on the sides
- Mic LED that can be turned on/off

### Generated Report

-

BorisBrock, 2023-05-25:
In the mean time I have done some analysis on the device. Here's how it is structured:

```
DEVICE ID 1b1c:0a6b on Bus 003 Address 013 =================
 bLength                :   0x12 (18 bytes)
 bDescriptorType        :    0x1 Device
 bcdUSB                 :  0x200 USB 2.0
 bDeviceClass           :    0x0 Specified at interface
 bDeviceSubClass        :    0x0
 bDeviceProtocol        :    0x0
 bMaxPacketSize0        :    0x8 (8 bytes)
 idVendor               : 0x1b1c
 idProduct              : 0x0a6b
 bcdDevice              :    0x0 Device 0.0
 iManufacturer          :    0x1 Corsair
 iProduct               :    0x2 CORSAIR HS80 RGB Wireless Gaming Receiver
 iSerialNumber          :    0x3 186752ca00030215
 bNumConfigurations     :    0x1
  CONFIGURATION 1: 500 mA ==================================
   bLength              :    0x9 (9 bytes)
   bDescriptorType      :    0x2 Configuration
   wTotalLength         :  0x1a3 (419 bytes)
   bNumInterfaces       :    0x5
   bConfigurationValue  :    0x1
   iConfiguration       :    0x0 
   bmAttributes         :   0xa0 Bus Powered, Remote Wakeup
   bMaxPower            :   0xfa (500 mA)
    INTERFACE 0: Audio =====================================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x0
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x0
     bInterfaceClass    :    0x1 Audio
     bInterfaceSubClass :    0x1
     bInterfaceProtocol :    0x0
     iInterface         :    0x0 
    INTERFACE 1: Audio =====================================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x1
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x0
     bInterfaceClass    :    0x1 Audio
     bInterfaceSubClass :    0x2
     bInterfaceProtocol :    0x0
     iInterface         :    0x0 
    INTERFACE 1, 1: Audio ==================================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x1
     bAlternateSetting  :    0x1
     bNumEndpoints      :    0x1
     bInterfaceClass    :    0x1 Audio
     bInterfaceSubClass :    0x2
     bInterfaceProtocol :    0x0
     iInterface         :    0x0 
      ENDPOINT 0x83: Isochronous IN ========================
       bLength          :    0x9 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x83 IN
       bmAttributes     :    0xd Isochronous
       wMaxPacketSize   :   0x62 (98 bytes)
       bInterval        :    0x1
    INTERFACE 1, 2: Audio ==================================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x1
     bAlternateSetting  :    0x2
     bNumEndpoints      :    0x1
[... 93 lines cut ...]
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x1
      ENDPOINT 0x2: Interrupt OUT ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x2 OUT
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x1
```

iCUE seems to send the color data to the endpoint 0x1 on interface 3. This is a package I captured:

```
# 02 09 06 00 09 00 00 00 FF 00 00 FF 00 00 FF 00    ........ÿ..ÿ€.ÿ.
#  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
#  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
#  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
```

The color data is sent as three bytes (R, G and B). In the above example they are all set to 0xFF (-> white).
The bytes after each 0xFF seem to be the color of the microphone LED, also 3 bytes (R, G and B). Here they are all 0x00 (-> LED off).

Maybe this helps with the implementation?

areiased, 2023-09-21:
Much needed support. I need to constantly boot into Windows to turn off the damn sides RGB since the battery with them on is literally 4-5h, while with them off is a whole week, and I'm not even joking. They get saved on the device (you have to use the "0% brightness" thing) but it eventually resets. Tip for HS80 users: This device doesn't seem to save RGB effects or colors on the device, its all iCUE dependent. But it does save the brightness setting under iCue device settings. Use it to make it 0%. But again, it eventually resets.

Being one of the best corsair headsets and arguably one of the best gaming headsets on the market, especially in terms of mic quality, and having a LOT of sales, there is an actual good amount of people out there on Linux which would love to have a proper way to control this device. We have two at home right now, and we both use Linux. 99% of the device already works perfectly under Linux. The only thing we need is a way to control (especially turn off) the RGB on the sides since they consume A LOT of battery.

I am on Nobara (Fedora-based), on a custom (modern and mostly high-end) full-AMD desktop. If I can do anything to help this development or to test out stuff, please tell me.

VBelozyorov, 2023-09-24:
+1 here, ready to check something on openSUSE Tumbleweed

clemperador, 2023-12-20:
+1, would love any workarounds at least for the mic led

ramonscode, 2024-06-10:
+1 please

ramonscode, 2024-06-10:
> Much needed support. I need to constantly boot into Windows to turn off the damn sides RGB since the battery with them on is literally 4-5h, while with them off is a whole week, and I'm not even joking. They get saved on the device (you have to use the "0% brightness" thing) but it eventually resets. Tip for HS80 users: This device doesn't seem to save RGB effects or colors on the device, its all iCUE dependent (which I hate). But it does save the brightness setting under iCue device settings. Use it to make it 0%. But again, it eventually resets.
> 
> The HS80 being one of the best corsair headsets and arguably one of the best gaming headsets on the market, especially in terms of mic quality, and having a LOT of sales, there is an actual good amount of people out there on Linux which would love to have a proper way to control this device. We have two at home right now, and we both use Linux. 99% of the device already works perfectly under Linux. The only thing we need is a way to control (especially turn off) the RGB on the sides since they consume A LOT of battery.
> 
> I am on Nobara (Fedora-based), on a custom (modern and mostly high-end) full-AMD desktop. If I can do anything to help this development or to test out stuff, please tell me.
> 
> Corsair should really do more iCue independent devices... Even Razer has on-board memory on all devices. Corsair is releasing more and more iCue dependent products so ckb-next importance is raising with literally every corsair release.

On Linux, have you managed to get the microphone to work with the headset? I am desperate. I'm using PopOS.

HenriqueBarucco, 2024-09-03:
I just wanted to turn off this damn mic feedback on linux...

JanEhehalt, 2025-07-12:
for anybody still trying to turn off the Headset LEDs, check this out: https://github.com/jurkovic-nikola/OpenLinkHub
Worked great for me :)

Kalle31p, 2026-01-21:
+1 support would be awesome