ckb-next/ckb-next #965: ckb-next not working

vrpsunited, 2023-05-17:
### Describe your situation in detail.

Hi all.
I decided to try to fix ckb-next as it is some weeks it doesn't work, basically it loads and then it doens't appear anywhere - there is no sign in the icon-tray.
It was working so I am sure my keyboard is compatible.

I have the Corsair K55 keybard, running on Arch Linux. I grabbed and built the version already from AUR repository as instructed.
I tried to reinstall but I couldn't because there were some files leftovers. I also asked for help in the irc channel already, a kind person told me to try to reinstall but forcing the overwrite which was ok for the installation process but then the program still doesnt run.

When on the terminal I run ckb-next, I get
`error while loading shared libraries: libquazip1-qt5.so.1.3.0: cannot open shared object file: No such file or directory`

They told me to try to rebuild the package as I did a pacman -Syu after installing it but I rebuilt it now and it doesn't work.
I also tried to look online and I ended up installing quazip which doens't help either.

Any other step to try to fix this issue?

### Additional Details

Add any additional details here.

### Generated Report (if applicable)

Generating report. Please wait...
        not a dynamic executable
Done! Please send the following report to the developers.
Location: /home/baruch/Programs/ckb-next/ckb-next-dev-detect-report.gz

aussyspitz, 2023-05-19:
I'm not sure if Arch is using the git version which I think you will need for the new K55 rgb pro keyboard this is the link to the ubuntu version https://launchpad.net/~tatokis/+archive/ubuntu/ckb-next-git

aussyspitz, 2023-05-19:
K55 | RGB | *Supported | 1b3d | ≤3.0
K55 | RGB PRO | *Supported | 1ba4 | **0.5.1 (git)**
K55 | RGB PRO LITE |  ????????  
K55 | RGB PRO XT | *Supported | 1ba1 | **0.5.1 (git)**

aussyspitz, 2023-05-19:
Just read you had it working before I don't have any experience with Arch I had a bit of trouble completely removing ckb-next but I basically searched the file system for anything related to ckb-next and manually deleted everything after I had run sudo apt purge ckb-next* in terminal and then added the ppa and installed the git version unfortunately that's probably not much help to you

gravityfargo, 2023-07-10:
No activity for a while and @aussyspitz gave a detailed response addressing all the concerns.

closed 2023-07-10 by gravityfargo

vrpsunited, 2023-07-31:
I was away for holidays. I am still having this problem sadly :(

gravityfargo, 2023-09-11:
> I am still having this problem sadly

> I was able to solve it

Is your install not working still?

MRNGSTAR, 2023-09-11:
It looks like he mistyped the command in terminal the first time "for some reason when I was writing s.1.3.0 it didn't work"
BUT then when he put in the correct command sudo ln -s /usr/lib/libquazip1-qt5.so.1.3 /usr/lib/libquazip1-qt5.so.1.0.0 he was able to solve it "I was able to solve it by doing sudo ln -s /usr/lib/libquazip1-qt5.so.1.3 /usr/lib/libquazip1-qt5.so.1.0.0"  his  comment was just worded awkwardly/backwards from what you would expect. I'm pretty sure you can mark this as completed