ckb-next/ckb-next #966: Installation impossible due to error downloading packages

VAX90, 2023-05-19:
### Describe your situation in detail.

When running the commands to download: sudo pacman -S base-devel cmake libsystemd qt5-base zlib libdbusmenu-qt5 quazip libpulse qt5-tools libxcb xcb-util-wm qt5-x11extras git

I get an error relative to this package [cmake-3.26.3-1-x86_64.pkg.tar.zst] resulting unavailable from every source in my mirrorlist. same thing obviusly happens with {yay ckb-next}.

i was able to git clone https://aur.archlinux.org/ckb-next.git but than i wouldent know if it would complile. Any advice on how to proceed without ./quickinstall ?

### Additional Details

Add any additional details here.

### Generated Report (if applicable)

Drag your generated report here if it's applicable.

Ravenslofty, 2023-05-19:
... When was the last time you ran `sudo pacman -Syu`?

aussyspitz, 2023-05-19:
can you manually install the newer version  https://archlinux.org/packages/extra/x86_64/cmake/

gravityfargo, 2023-07-10:
This issue is stale and not directly related to ckb-next so I'm going to close it. The error is most likely the archlinux-keyring package being out of date.

closed 2023-07-10 by gravityfargo