ckb-next/ckb-next #970: k70 pro mini wireless

luthiwd, 2023-06-07:
### What kind of device is it?

Keyboard

### What is the name of the device?

k70 pro mini wireless

### Link to Corsair's Product Page

https://www.corsair.com/es/es/p/keyboards/ch-9189010-na/k70-pro-mini-wireless-60-mechanical-cherry-mx-red-switch-keyboard-with-rgb-backlighting-black-ch-9189010-na

### Additional Details

Add any additional details here.

### Generated Report

[ckb-next-report-1686118146.gz](https://github.com/ckb-next/ckb-next/files/11673130/ckb-next-report-1686118146.gz)

gravityfargo, 2023-09-12:
Could you please remove the wireless dongle, plug the keyboard in via usb and regenerate the report?

luthiwd, 2023-09-13:
Yes!!!

padrino121, 2023-09-18:
Attaching a report with it plugged in via USB
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/12644534/ckb-next-dev-detect-report.gz)

luthiwd, 2023-09-19:
[ckb-next-report-1695104480.gz](https://github.com/ckb-next/ckb-next/files/12656486/ckb-next-report-1695104480.gz)

mklein994, 2024-01-05:
I experimented with adding support for this, and I managed to get most of the lights to work, but since I'm brand-new to this project (and C++ in general), I'm not really sure what I'm doing. 🙃

I'm mostly interested in adding RGB support, so I started off by essentially copying[^1] what was done for the [K65](https://www.corsair.com/ca/en/p/keyboards/ch-9194014-na/k65-rgb-mini-60-mechanical-gaming-keyboard-cherry-mx-speed-black-ch-9194014-na), since they basically have the same layout. One major part that's still missing is the light strip (i.e. the "RGB LightEdge"), but I'm not sure how to add that.

Report collected with USB cable attached: [ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/13844976/ckb-next-dev-detect-report.gz)

[^1]: Sorry, don't have any patches yet.