ckb-next/ckb-next #972: K70 RGB MK.2 - LED brightness issue after upgrade to 0.6.0

TheTee82, 2023-06-23:
### Affected Devices

K70 RGB MK.2 LP

### Linux Distribution

Arch

### Installation Method

AUR

### ckb-next version

ckb-next 0.6.0-8-g6c44fd9e

### What did you expect?

After upgrade from `0.5.x` to `0.6.0`:

When switching through LED-Brightness-Button brightness shall be changed - as was previously working.
However, when cycling through brightness (either hardware button or dropdown menu in ckb-next client), lighting changes as if cycling through profiles (however, I do not have profiles with these colors).

"All off" does not work at all for example but switches to a configuration I don't have (afaict). Also, only the "33%" brightness setting actually lights up the correct LEDs.

### Additional Details

Tried with both `ckb-next 0.6.0-1` and uninstalling and installing `ckb-next-git 1:0.6.0.r0.gfaca39b1-1` from AUR

### Generated Report

[ckb-next-report-1687527685.gz](https://github.com/ckb-next/ckb-next/files/11849009/ckb-next-report-1687527685.gz)

tatokis, 2023-06-23:
Does this happen in the keyboard preview in the application, or just on the keyboard?
Would you also feel comfortable with sharing the contents of `~/.config/ckb-next` ? It's where all the config and profiles are stored.

TheTee82, 2023-06-23:
>Does this happen in the keyboard preview in the application too,

It's just on the keyboard - preview looks fine.
And I stand corrected: for some it's on 33% OK but others seem to be behaving weird at different brightness settings.

---

>Would you also feel comfortable with sharing the contents of `~/.config/ckb-next`

Sure. However, there's just the `ckb-next.conf` file.
[ckb-next.conf.tar.gz](https://github.com/ckb-next/ckb-next/files/11849526/ckb-next.conf.tar.gz)

For example in profile 6 (? according to config file I guess) is as follows: 
- Lighting: 100% -> all "lit" keys are just bright red
- Lighting: 67% -> actually OK
- Lighting: 33% -> OK as well
- Lighting: off -> white keys (conf) are now cyan, orange and yellow keys (conf) are now green

tatokis, 2023-06-23:
I can't reproduce this no matter what I try. (Granted, with a different keyboard.)

Can you try manually building and running the daemon from 0.5.0 while running the GUI from your system wide 0.6.0/git installation?

Should be a matter of:
1. leave the gui running from your system-wide installation and make sure it's `>= 0.6.0`.
2. Obtain the source for `0.5.0` and `cd` into the directory
3. `mkdir build && cd build`
4. `cmake .. -DCMAKE_BUILD_TYPE=Debug`
5. `make daemon -j4`
6. `sudo systemctl stop ckb-next-daemon`
7. `sudo ./bin/ckb-next-daemon`

If running the old daemon with the new gui fixes it, can you please do a `git bisect`?
If you want any additional help, feel free to join the IRC/Matrix chat. Links are in the readme.

TheTee82, 2023-06-23:
Solved by unplugging the keyboard from USB-Ports and plugging it back in again.
Seems to be stable now.

Thank you so much for your time and support in IRC!

closed 2023-06-23 by TheTee82