ckb-next/ckb-next #973: K95 RGB PLATINUM - various issues after upgrade to 0.6.0

G4Zz0L1, 2023-06-24:
### Affected Devices

K95 RGB PLATINUM

### Linux Distribution

Arch Linux

### Installation Method

AUR

### ckb-next version

ckb-next 0.6.0

### What did you expect?

I can't change the brightness with the apposite button, or via gui. Even changins brightness on a single key it's not working.
After boot, the keyboard remains shut down, it light up only when I press a key.
It seems it's not responding to the profile switch command, it starts only when I press a key, as before.
The previous version was working fine, all the animations behaved correctly and the profile switching was working.
Can you point me to something to check or to debug?
Thanks in advance!

### Did you try anything that may have helped?

I've cleared the config folder, disconnected and reconnected the keyboard, even several reboots in between.

### Additional Details

Add any additional details here.

### Generated Report

[ckb-next-report-1687623738.gz](https://github.com/ckb-next/ckb-next/files/11857654/ckb-next-report-1687623738.gz)

G4Zz0L1, 2023-10-05:
Any news on this?

tatokis, 2023-10-26:
You need to lower the FPS in the GUI. Do you have it set to 60? Before 0.6.0, 60 FPS didn't actually run at 60, whereas now it does, and some setups can not handle it.

G4Zz0L1, 2023-10-26:
I'm already at 30 fps, but it doesn't work anyway.
I've tried everything from 15 to 60, same result for everything.

G4Zz0L1, 2023-12-18:
The problem fixed itself, without upgrading ckb-next, so I suppose it's some external dependency or library, but I can't guess which it is.
Now everything listed above works as expected on version 0.6.0, even the 60 fps.

closed 2023-12-18 by G4Zz0L1