ckb-next/ckb-next #974: limited inputs/sec speed, when lighting frame rate is 60 (after update?)

ana-rchy, 2023-06-26:
### Affected Devices

corsair strafe rgb v1

### Linux Distribution

arch

### Installation Method

aur (non-git version)

### ckb-next version

0.6.0

### What did you expect?

unlimited typing speed, instead you can only get like 3-4 inputs happening per ~0.5 secs until the next "batch" can come through (the inputs dont seem to be buffered, tho)

### Did you try anything that may have helped?

reinstalled all of arch lmao

### Additional Details

didnt have this issue before, assumedly before the 0.6.0 update but im not sure, when i for sure had the lighting set to 60fps

### Generated Report

[ckb-next-report-1687788761.gz](https://github.com/ckb-next/ckb-next/files/11870273/ckb-next-report-1687788761.gz)

tatokis, 2023-06-26:
Previously, 60FPS wasn't actually 60 because the timing calculations were off and made incorrect assumptions. If I recall correctly it actually ran at 50 instead.

0.6.0 fixed it and it actually runs at 60 now. It's possible the keyboard doesn't like it when connected to the USB controller your system is using.

That said, I can not reproduce this at 60FPS on a relatively slow machine.

If it runs correctly at 50FPS then it's not really a regression, but more a sideffect of actually running at 60FPS. That said, there could be a bug in the code that causes this at higher framerates now, but only got exposed after allowing it to run at the advertised speed.

Does it work at 50?

ana-rchy, 2023-06-26:
> Previously, 60FPS wasn't actually 60 because the timing calculations were off and made incorrect assumptions. If I recall correctly it actually ran at 50 instead.
>
> 0.6.0 fixed it and it actually [...]

it does work at 50, or at the very least if there is some limiting going on its unnoticeable

pointydev, 2023-07-04:
After upgrading to 0.6.0, I also experienced limited inputs for keystrokes on the same key within a short time (e.g. `hello` would result in `helo`, `applications` would result in `aplications`, pressing backspace 5 times might only result in 2). While I do use 60 FPS animations, I can't confirm if this behaviour was limited to it.
```
OS: Fedora release 38 (Thirty Eight) x86_64 
Kernel: 6.3.8-200.fc38.x86_64 
DE: Plasma 5.27.6 
WM: KWin 
```
I ended up downgrading back to 0.5.0 to fix this issue, but a temporary workaround was to use the "BIOS mode" switch on my keyboard (K70 LUX RGB).

paypur, 2023-07-06:
I'm experiencing the same issue as well on ckb-next 6.0.0 on a corsair strafe rgb. I seem to have resolved it by changing the polling rate