ckb-next/ckb-next #978: Enabling experimental mode detects Dark Core RGB Pro SE but extra buttons no longer work

sam-k-taylor, 2023-07-05:
### Affected Devices

Dark Core RGB Pro SE

### Linux Distribution

Ubuntu 22.04

### Installation Method

Build from source

### ckb-next version

ckb-next v0.6.0

### What did you expect?

When enabling experimental mode Dark Core RGB Pro SE is detected but extra buttons(Back, Forward, Profile Cycle, ect...) no longer work

### Additional Details

Add any additional details here.

### Generated Report

[ckb-next-report-1688545778.gz](https://github.com/ckb-next/ckb-next/files/11955710/ckb-next-report-1688545778.gz)

sam-k-taylor, 2023-07-17:
Is this project dead?

sam-k-taylor, 2023-08-03:
additional info after talking to KitsuWhooa on IRC

[image]

[image]

[image]

https://pastebin.com/ZJXQmYe5

they had me rebui0ld with the following options
```
cmake .. -DCMAKE_BUILD_TYPE=Debug -DDEBUG_USB_INPUT=1
make daemon -j4
```

Not seeing any diifferent output when running  `sudo ckb-next-daemon --enable-experimental`

olokos, 2024-06-07:
It is now functional with current code, albeit daemon still needs `--enable-experimental`. 
Check my report in #575  - almost everything works now, but You need to compile ckb-next Yourself, latest stable release is still bugged.

I think this issue can be closed.

closed 2025-05-31 by tatokis