ckb-next/ckb-next #980: UI not showing on ubuntu 22.04 LTS

sassouiabd, 2023-07-07:
### Affected Devices

UI

### Linux Distribution

Ubuntu 22.04.2 LTS

### Installation Method

I have tried both from apt-get install and also building the project

### ckb-next version

ckb-next 0.6.0-20-g932ff43

### What did you expect?

I except the UI to show when i launch ckb-next and instead i get a dark screen
[image]

### Did you try anything that may have helped?

When launching ckb-next from terminal i got those error, i don't know if it may help or it just warning : 
qt.qpa.backingstore: composeAndFlush: QOpenGLContext creation failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.gui.icc: Unsupported ICC profile class 70727472
QPngHandler: Failed to parse ICC profile
qt.gui.icc: Unsupported ICC profile class 70727472
QPngHandler: Failed to parse ICC profile
qt.gui.icc: Unsupported ICC profile class 70727472
QPngHandler: Failed to parse ICC profile
qt.gui.icc: Unsupported ICC profile class 70727472
QPngHandler: Failed to parse ICC profile
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: QOpenGLContext creation failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.gui.icc: Unsupported ICC profile class 70727472
QPngHandler: Failed to parse ICC profile
qt.gui.icc: Unsupported ICC profile class 70727472
QPngHandler: Failed to parse ICC profile
qt.gui.icc: Unsupported ICC profile class 70727472
QPngHandler: Failed to parse ICC profile
qt.gui.icc: Unsupported ICC profile class 70727472
QPngHandler: Failed to parse ICC profile
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed

### Additional Details

Add any additional details here.

### Generated Report

the user interface is not available.

Ravenslofty, 2023-07-07:
Are you using Nvidia graphics?

sassouiabd, 2023-07-07:
Yes, here is my config : 
00:00.0 Host bridge: Intel Corporation 4th Gen Core Processor DRAM Controller (rev 06)
00:01.0 PCI bridge: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor PCI Express x16 Controller (rev 06)
00:14.0 USB controller: Intel Corporation 9 Series Chipset Family USB xHCI Controller
00:16.0 Communication controller: Intel Corporation 9 Series Chipset Family ME Interface #1
00:19.0 Ethernet controller: Intel Corporation Ethernet Connection (2) I218-V
00:1a.0 USB controller: Intel Corporation 9 Series Chipset Family USB EHCI Controller #2
00:1b.0 Audio device: Intel Corporation 9 Series Chipset Family HD Audio Controller
00:1c.0 PCI bridge: Intel Corporation 9 Series Chipset Family PCI Express Root Port 1 (rev d0)
00:1c.3 PCI bridge: Intel Corporation 82801 PCI Bridge (rev d0)
00:1d.0 USB controller: Intel Corporation 9 Series Chipset Family USB EHCI Controller #1
00:1f.0 ISA bridge: Intel Corporation Z97 Chipset LPC Controller
00:1f.2 SATA controller: Intel Corporation 9 Series Chipset Family SATA Controller [AHCI Mode]
00:1f.3 SMBus: Intel Corporation 9 Series Chipset Family SMBus Controller
01:00.0 VGA compatible controller: NVIDIA Corporation GM204 [GeForce GTX 970] (rev a1)
01:00.1 Audio device: NVIDIA Corporation GM204 High Definition Audio Controller (rev a1)
03:00.0 PCI bridge: ASMedia Technology Inc. ASM1083/1085 PCIe to PCI Bridge (rev 04)

sassouiabd, 2023-07-08:
It seems that the issue was not related to ckb.
I had reinstalled my graphical drivers and the problem have been solved.
Thanks @Ravenslofty

closed 2023-07-08 by sassouiabd