ckb-next/ckb-next #990: CORSAIR K70 RGB PRO Optical-Mechanical Gaming Keyboard 1b1c:1bc6

TechZ-S, 2023-08-04:
### What kind of device is it?

Keyboard

### What is the name of the device?

K70 RGB PRO Optical-Mechanical

### Additional Details

I could not find any issue regarding the K70 RGB PRO Optical-Mechanical with USB ID 1b1c:1bc6

### Generated Report

Drag your generated report here.
[ckb-next-report-1691149967.gz](https://github.com/ckb-next/ckb-next/files/12260414/ckb-next-report-1691149967.gz)

Code-Slave, 2023-08-04:
heres my report too

[ckb-next-report-1691189970.gz](https://github.com/ckb-next/ckb-next/files/12265069/ckb-next-report-1691189970.gz)

PCMartinzzon, 2023-08-25:
[ckb-next-report-1692988925.gz](https://github.com/ckb-next/ckb-next/files/12442717/ckb-next-report-1692988925.gz)

wernerm, 2023-09-11:
Is driver support really necessary for this keyboard model? 

I have received and started to use one recently and it offers 90% of what I wanted out-of-the-box, with on-board support. (It's a great keyboard.)

You might want to give [the manual](https://www.manualslib.com/manual/2894314/Corsair-K70-Rgb-Pro.html) just three minutes of your time - there's a generous amount of customising that can be done using only the hardware.

A bit of background: I am basically a linux-only user that used to own a K70 RGB Lux (until it finally gave in). The ckb-next driver has offered excellent support for that keyboard - many thanks to this great project.  I recently got a replacement keyboard, the K70 RGB PRO. It offers on-board storage and built-in macro support. there's a lot you can do without even having a driver. To be honest, even if I was a Windows user, I would probably not install iCue due to the bloatware. Well, it's not important for my personal needs of course... I am a coder, so some serious gamers will obviously want better support that that offered by hardware - but if you are a serious gamer, then one can assume that you are probably not using Linux *exclusively* for you gaming. Why not then boot up a Windows VM, install iCue, set up your keyboard profiles (and so on), save it to the 'board and return to Linux with your customised beast? I mean, really, how often do you customise your keyboard?