ckb-next/ckb-next #992: Please add support for iCUE H100i Elite CAPELLIX XT

kindagonzo, 2023-08-23:
### What kind of device is it?

AIO Coolers

### What is the name of the device?

Corsair iCUE H100i Elite CAPELLIX XT

### Link to Corsair's Product Page

https://www.corsair.com/us/en/p/cpu-coolers/cw-9060068-ww/icue-h100i-elite-capellix-xt-liquid-cpu-cooler

### Additional Details

Add any additional details here.

### Generated Report

Drag your generated report here.

gravityfargo, 2023-09-11:
With the exception of the older H115i AIOs (and probably a few more), most of the pump All-in-Ones are controlled by the Commander Core or Commander Pro, which support has been requested for. By the product page, this one uses the Commander Core.

 See: https://github.com/ckb-next/ckb-next/issues/833

closed 2024-04-25 by gravityfargo

kindagonzo, 2024-04-25:
@gravityfargo thank you!