ckb-next/ckb-next #993: Add support for SCIMITAR ELITE WIRELESS MMO Gaming Mouse

lohvht, 2023-09-09:
### What kind of device is it?

Mouse

### What is the name of the device?

SCIMITAR ELITE WIRELESS MMO Gaming Mouse

### Link to Corsair's Product Page

https://www.corsair.com/us/en/p/gaming-mouse/ch-9314311-na/scimitar-elite-wireless-mmo-gaming-mouse-ch-9314311-na

### Additional Details

Add any additional details here.

### Generated Report

Drag your generated report here.
[ckb-next-report-1694249968.gz](https://github.com/ckb-next/ckb-next/files/12565358/ckb-next-report-1694249968.gz)

gilvbp, 2023-09-19:
@lohvht Hi there, could you dump without cable, just donkey plugged?

lohvht, 2023-09-21:
Hi! Thanks for responding, certainly!

# Additional Details

Add any additional details here.

The output from the commands below are dumped with mouse connected via the slipstream dongle.

```bash
❯ sudo ./getUsbString 0x1b1c 0x1bdc 4
Read 52 bytes
Raw (hex): 34 03 42 00 6f 00 6f 00 74 00 6c 00 6f 00 61 00 64 00 65 00 72 00 20 00 76 00 65 00 72 00 73 00 69 00 6f 00 6e 00 3a 00 20 00 30 00 2e 00 30 00 2e 00 36 00 
String: Bootloader version: 0.0.6

❯ sudo ./getUsbString 0x1b1c 0x1bdc 5
Read 10 bytes
Raw (hex): 0a 03 42 00 50 00 30 00 30 00 
String: BP00

sudo usbhid-dump -d 1b1c:1bdc
003:008:003:DESCRIPTOR         1695303164.191031
 05 01 09 06 A1 01 85 01 05 08 95 03 75 01 19 01
 29 03 15 00 25 01 91 02 95 05 91 03 05 07 75 01
 19 E0 29 E7 15 00 25 01 95 08 81 02 19 00 29 67
 15 00 25 01 95 68 81 02 19 87 29 8E 95 08 81 02
 C0 05 0C 09 01 A1 01 85 02 95 01 75 10 15 00 26
 FF 03 19 00 2A FF 03 81 00 C0

003:008:004:DESCRIPTOR         1695303164.194015
 05 01 09 02 A1 01 09 01 A1 00 75 10 95 02 05 01
 09 30 09 31 16 01 80 26 FF 7F 81 06 C0 C0

003:008:005:DESCRIPTOR         1695303164.197015
 05 01 09 06 A1 01 85 01 05 07 75 01 19 E0 29 E7
 15 00 25 01 95 08 81 02 19 00 29 8B 15 00 26 FF
 00 75 08 95 0D 81 00 C0 05 0C 09 01 A1 01 85 02
 95 01 75 10 15 00 26 FF 03 19 00 2A FF 03 81 00
 C0

003:008:002:DESCRIPTOR         1695303164.200004
 06 42 FF 09 02 A1 01 15 00 26 FF 00 75 08 95 40
 09 02 81 02 C0

003:008:001:DESCRIPTOR         1695303164.203008
 06 42 FF 09 01 A1 01 15 00 26 FF 00 75 08 95 40
 09 01 81 02 95 40 09 01 91 02 C0

003:008:000:DESCRIPTOR         1695303164.206006
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 20
 15 00 25 01 95 20 75 01 81 02 75 10 95 02 05 01
 09 30 09 31 16 01 80 26 FF 7F 81 06 75 08 95 01
 05 01 09 38 15 81 25 7F 81 06 05 0C 0A 38 02 95
 01 81 06 75 08 95 05 06 00 FF 09 F1 15 00 26 FF
 00 81 00 C0 C0
```

# Generated Report

Drag your generated report here.

The report below is dumped with mouse connected via the dongle
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/12685160/ckb-next-dev-detect-report.gz)

gilvbp, 2023-09-30:
@lohvht hi there. You can use my branch https://github.com/ckb-next/ckb-next/pull/997 to test your mouse. Use these instructions: https://github.com/ckb-next/ckb-next/pull/855#issuecomment-1178343114

INTOODEEP-dev, 2024-01-08:
> @lohvht hi there. You can use my branch #997 to test your mouse. Use these instructions: [#855 (comment)](https://github.com/ckb-next/ckb-next/pull/855#issuecomment-1178343114)

I really want to get this working for my Scimitar Elite Wireless, and I thank you very much for putting the work in to address the issue, but I cannot figure out how to use your branch. I am admittedly a big idiot when it comes to things like this. 

I am currently using the ckb-next AUR package on arch. I tried adding that experimental option in the #855 link, but it didn't seem to change anything for me. Am I able to just switch something around to add the fixes you made, or do I have to like compile it from github? I've never done anything like that before and I see the wiki page on how to do compile from source, but I'm confused as to how to compile your specific pull request. 

Sorry if this is all basic and I'm over complicating things. Tried to figure this out on my own before asking this, but just couldn't figure it out.

gilvbp, 2024-01-08:
> > @lohvht hi there. You can use my branch #997 to test your mouse. Use these instructions: [#855 (comment)](https://github.com/ckb-next/ckb-next/pull/855#issuecomment-1178343114)
> 
> I really want to get this working for my Scimitar Elite Wireless, and I thank you very much for putting the work in to address the issue, but I cannot figure out how to use your branch. I am admittedly a big idiot when it comes to things like this. 
> 
> I am currently using the ckb-next AUR package on you. I tried adding that experimental option in the #855 link, but it didn't seem to change anything for me. Am I able to just switch something around to add the fixes you made, or do I have to like compile it from github? I've never done anything like that before and I see the wiki page on how to do compile from source, but I'm confused as to how to compile your specific pull request. 
> 
> Sorry if this is all basic and I'm over complicating things. Tried to figure this out on my own before asking this, but just couldn't figure it out. 

You're welcome! Yes, first you have to checkout my branch, delete AUR package, then run installation steps (https://github.com/ckb-next/ckb-next/wiki/Linux-Installation), and after installation it's completed, you can add experimental parameters on you daemon.service, as you already look at old @Clofriwen post).

You don't have to apologize, I'm here to help.

You can send me an email if got stuck on that, OK?

cirk2, 2024-01-09:
Did a short test with my device:
So far seems to work really good. Keybinding and LED Animations work.

Only things I noticed so far is the DPI Indicator only doing Green an Yellow tones as if the green channel is stuck on 255 and Blue channel on 0. The other thing is the device DPI config itself I can not notice a difference between 800 and 6000 dpi, which seems wrong.

gilvbp, 2024-01-09:
> Did a short test with my device: So far seems to work really good. Keybinding and LED Animations work.
>
> Only things I noticed so far is the DPI Indicator only doing Green an Yellow tones as if the [...]

Hi cirk2, thanks for testing! I will check the led dpi configuration. Yes, changing DPI with the button only works when the CKB program GUI is not running, only on daemon (the button is mapped correctly on GUI), my assumption is that the wireless version uses different DPI actions besides the wired one. That's one thing that I'm working on right now to fix.

gilvbp, 2024-02-05:
> Did a short test with my device: So far seems to work really good. Keybinding and LED Animations work.
>
> Only things I noticed so far is the DPI Indicator only doing Green an Yellow tones as if the [...]

Hi there, I fixed the 12 button and the dpi action values, @INTOODEEP-dev, check my code again.

cirk2, 2024-02-06:
So far from my tests:
DPI xhange works fine now.
Button 12 Binding also works well.

The DPI indicator LED colors are still wrong (white and red both make the led yellow).

gilvbp, 2024-02-06:
> So far from my tests:
> DPI xhange works fine now.
> Button 12 Binding also works well.
> 
> The DPI indicator LED colors are still wrong (white and red both make the led yellow).

Yes, but there are fewer problems, one bug that's still remains its the side mouse button macros doesn't work, work only on keyboard. I'll see that issue with the ex-manteiner on irc channel.

Martan404, 2024-05-12:
Here's my report in case it helps
[ckb-next-report-1715529666.gz](https://github.com/ckb-next/ckb-next/files/15286916/ckb-next-report-1715529666.gz)

AQDuck97, 2024-06-26:
> @lohvht hi there. You can use my branch #997 to test your mouse. Use these instructions: [#855 (comment)](https://github.com/ckb-next/ckb-next/pull/855#issuecomment-1178343114)

I tried your fork and I'm able to rebind as I want, but the buttons don't do anything at all outside of CKB (they work if I disable the daemon, though will just be the numeric row)

AQDuck97, 2024-06-28:
@gilvbp I've run into a couple issues with your fork, if I leave the daemon running but disconnect the mouse in some way (run out of battery, suspend the system, plug in the cable, etc.) the daemon will freeze up and not able to deactivate or detect the mouse again once I turn the mouse back on, forcing me to reboot the entire PC.

Some smaller issues I have is I can't change the RGB of the numpad or the daemon freezes up again (logo RGB works fine) and "forward" button is button 5 but it doesn't work at all on webbrowsers or in games, button 6 acts more like "forward" in browsers but also not recognized in games.

IMidULti, 2024-09-30:
I don't have the freezing issue described by @AQDuck97, almost everything works except assigned Macros and the assignment of the "5 (Forward)" special button.

cirk2, 2024-10-07:
I only get the freezing of the daemon when disconnecting the wired connection, wireless disconnects are fine.
Restarting the daemon (need a more forceful signal like `killall -9 ckb-next-daemon`).

One other thing: The mouse wheel behaves a bit strange. Scrolling fast makes it skip some "clicks", not scrolling as fast as expected, scrolling fast enough makes it don't scroll at all or even scroll backwards. Having the mouse connected wireless and wired (i.e. when charging) makes this noticeable worse, as if the events are doubled.

DreiPixel, 2024-10-18:
Same issues as above. Freezing daemon, macros not working, and the 5 (Forward) button.
everything else seems to work good.

tuxx, 2024-12-01:
Seems to work pretty good. The only issue i have found so far is the Forward special button is not working.

tuxx, 2024-12-01:
Added a pull request to fix forward button: https://github.com/gilvbp/ckb-next/pull/4

gilvbp, 2024-12-02:
> Added a pull request to fix forward button: https://github.com/gilvbp/ckb-next/pull/4

Merged. Thanks

IMidULti, 2024-12-19:
@tuxx thanks for fixing that, forward works for me now. The only issue left is the macros.

nilbest, 2025-04-06:
Hey Guys, I'm not shure why, but I bought the SCIMITAR RGB ELITE WIRELESS Gaming Mouse last Week and it will not work with ckb-next.
Is this a known Issue?

Here is my lsusb output.
So the device may be the correct one, but it never showed up as a detected device in ckb-next.
```
lsusb -d 1b1c:
Bus 001 Device 003: ID 1b1c:0a6b Corsair CORSAIR HS80 RGB Wireless Gaming Receiver
Bus 001 Device 005: ID 1b1c:1bdc Corsair CORSAIR SLIPSTREAM WIRELESS USB Receiver
Bus 001 Device 013: ID 1b1c:1bdb Corsair CORSAIR SCIMITAR RGB ELITE WIRELESS Gaming Mouse
```

gilvbp, 2025-04-06:
> Hey Guys, I'm not shure why, but I bought the SCIMITAR RGB ELITE WIRELESS Gaming Mouse last Week and it will not work with ckb-next. Do you know if this is still a known Issue?
> 
> Here is my lsusb output. So the device may be the correct one, but it never showed up as a detected device in ckb-next. At least it works with basic functions. Basic RGB Setup and the 12 number keys work like number keys. But I can not change anything.
> 
> ```
> lsusb -d 1b1c:
> Bus 001 Device 003: ID 1b1c:0a6b Corsair CORSAIR HS80 RGB Wireless Gaming Receiver
> Bus 001 Device 005: ID 1b1c:1bdc Corsair CORSAIR SLIPSTREAM WIRELESS USB Receiver
> Bus 001 Device 013: ID 1b1c:1bdb Corsair CORSAIR SCIMITAR RGB ELITE WIRELESS Gaming Mouse
> ```
> 
> Or does your fork work? [@gilvbp](https://github.com/gilvbp) I haven't tied it yet.

Yes, it's working. Only macros still not working.

nilbest, 2025-04-06:
So I do not have to enable the experimental settings anymore?

nilbest, 2025-04-06:
> Yes, it's working. Only macros still not working. You can use my main branch.

So I do not have to enable the experimental settings anymore?
As mentioned in https://github.com/ckb-next/ckb-next/pull/855#issuecomment-1178343114 ?

gilvbp, 2025-04-07:
> > Yes, it's working. Only macros still not working. You can use my main branch.
> 
> So I do not have to enable the experimental settings anymore? As mentioned in [#855 (comment)](https://github.com/ckb-next/ckb-next/pull/855#issuecomment-1178343114) ?
> 
> Edit: Yes you have to. Enable them right now.

Yes, it still needed.

Martan404, 2026-10-06:
What would be required to get macros working so support can get added to main?