ckb-next/ckb-next #1000: ckb-next segfaults on Arch Linux Wayland (Sway)

thumsl, 2023-10-08:
### Affected Devices

happens whether I connect USB dongle or not

### Linux Distribution

Arch Linux

### Installation Method

AUR ckb-next-git (built from source)

### ckb-next version

ckb-next 0.6.0-29-gb2188fac

### What did you expect?

ckb-next launches and I can configure devices but program segfaults

segfaults instanly if daemon is running
segfaults after pressing ok on the "daemon not running" alert when daemon is not running

### Did you try anything that may have helped?

tried running ckb-next with the daemon running, with the daemon not running, with the daemon running with experimental flag and without the mouse dongle being connected, no difference

I suspect it has something to do with wayland/sway or some dependecy missing?

How can I help debug?

Thanks

### Additional Details

console output when running ckb-next:

ckb-next 0.6.0-29-gb2188fac
Using "AMD" "AMD Radeon RX 6700 XT (navi22, LLVM 16.0.6, DRM 3.54, 6.5.5-arch1-1)"
QObject::startTimer: Timers cannot have negative intervals
Path to settings: "/home/luancristian/.config/ckb-next/ckb-next.conf"
Scanning "/usr/lib/ckb-next-animations/gradient"
Scanning "/usr/lib/ckb-next-animations/heat"
Scanning "/usr/lib/ckb-next-animations/invaders"
Scanning "/usr/lib/ckb-next-animations/life"
Scanning "/usr/lib/ckb-next-animations/mviz"
Scanning "/usr/lib/ckb-next-animations/pinwheel"
Scanning "/usr/lib/ckb-next-animations/pipe"
Scanning "/usr/lib/ckb-next-animations/rain"
Scanning "/usr/lib/ckb-next-animations/random"
Scanning "/usr/lib/ckb-next-animations/ripple"
Scanning "/usr/lib/ckb-next-animations/snake"
Scanning "/usr/lib/ckb-next-animations/wave"
[1]    52101 segmentation fault (core dumped)  ckb-next`

### Generated Report

Drag your generated report here.
ckb-next-dev-detect - Sun Oct  8 20:53:00 -03 2023

```
Kernel: Linux Ryzen9 6.5.5-arch1-1 #1 SMP PREEMPT_DYNAMIC Sat, 23 Sep 2023 22:55:13 +0000 x86_64 GNU/Linux

OS Version:
LSB Version:	n/a
Distributor ID:	Arch
Description:	Arch Linux
Release:	rolling
Codename:	n/a

[... 383 lines cut ...]
Oct 05 20:53:28.668732 Ryzen9 ckb-next-daemon[1656]: [I] Removed device path /dev/input/ckb0
Oct 05 20:53:28.668634 Ryzen9 systemd[1]: Stopping Corsair Keyboards and Mice Daemon...
Oct 05 20:53:28.668953 Ryzen9 systemd[1]: ckb-next-daemon.service: Deactivated successfully.
Oct 05 20:53:28.734144 Ryzen9 systemd[1]: Stopped Corsair Keyboards and Mice Daemon.
Oct 05 20:55:06.900926 Ryzen9 systemd[1]: Started Corsair Keyboards and Mice Daemon.
Oct 05 20:55:06.901606 Ryzen9 ckb-next-daemon[10023]: ckb-next-daemon 0.6.0-29-gb2188fac
Oct 05 20:55:06.901876 Ryzen9 ckb-next-daemon[10023]: [I] Root controller ready at /dev/input/ckb0
Oct 05 20:55:13.377447 Ryzen9 ckb-next-daemon[10023]: [I] Caught signal 15
Oct 05 20:55:13.377447 Ryzen9 ckb-next-daemon[10023]: [I] Closing root controller
Oct 05 20:55:13.377616 Ryzen9 ckb-next-daemon[10023]: [I] Removed device path /dev/input/ckb0
Oct 05 20:55:13.377530 Ryzen9 systemd[1]: Stopping Corsair Keyboards and Mice Daemon...
Oct 05 20:55:13.378262 Ryzen9 systemd[1]: ckb-next-daemon.service: Deactivated successfully.
Oct 05 20:55:13.407464 Ryzen9 systemd[1]: Stopped Corsair Keyboards and Mice Daemon.

Dev nodes:
find: '/dev/input/ckb*': No such file or directory

Kernel command line:
cryptdevice=UUID=9d12750d-1837-4995-8a9f-3a6191d0427f:root root=/dev/mapper/root rw add_efi_memmap initrd=amd-ucode.img initrd=initramfs-linux.img

GUI ldd:

Environment:

```

tatokis, 2023-10-09:
Assuming you have it installed, what does `coredumpctl dump ckb-next` say?

thumsl, 2023-10-09:
> Assuming you have it installed, what does `coredumpctl dump ckb-next` say?

 I get "no match found" informing ckb-next, but if I just run `coredumpctl dump` I get the dump from ckb-next:
 
       Message: Process 59394 (ckb-next.real) of user 1000 dumped core.
 
                Stack trace of thread 59394:
                #0  0x00007f4cf4b38b68 _ZNK15QtWaylandClient32QWaylandWindowManagerIntegration16showIsFullScreenEv (libQt5WaylandClient.so.5 + 0x88b68)
                #1  0x00007f4cf4b1a944 _ZNK15QtWaylandClient19QWaylandIntegration9styleHintEN20QPlatformIntegration9StyleHintE (libQt5WaylandClient.so.5 + 0x6a944)
                #2  0x00007f4cf992fd0d _ZNK20QPlatformIntegration18defaultWindowStateE6QFlagsIN2Qt10WindowTypeEE (libQt5Gui.so.5 + 0x12fd0d)
                #3  0x00007f4cfa1aa1e2 _ZN7QWidget4showEv (libQt5Widgets.so.5 + 0x1aa1e2)
                #4  0x0000561621b396a6 main (ckb-next.real + 0x576a6)
                #5  0x00007f4cf8c45cd0 n/a (libc.so.6 + 0x27cd0)
                #6  0x00007f4cf8c45d8a __libc_start_main (libc.so.6 + 0x27d8a)
                #7  0x0000561621b39c85 n/a (ckb-next.real + 0x57c85)

thumsl, 2023-10-09:
I did not have Xwayland running and that's why it wasn't working (realized when Steam wouldnt open either).

**However I had Xwayland installed, and the sway documentation says that it's enabled by default however it only really worked after I explicitly added "xwayland enable" to the sway config file and rebooted**

Should this program require Xwayland or is it a bug that still needs to be investigated?

Probably same as #764

commit c93e9510 (2023-10-26): gui: Add error handling for xcb_connect

This is needed because xcb_connect returns a valid value even if it fails.
Therefore we need to use xcb_connection_has_error() to check for failure.

Resolves #1000
PR #1005 (2023-10-26): gui: Add error handling for xcb_connect
This is needed because xcb_connect returns a valid value even if it fails.
Therefore we need to use xcb_connection_has_error() to check for failure.

Resolves #1000
closed 2023-10-26 by tatokis