ckb-next/ckb-next #1002: H150iELITE doesn't seem to show up

xana43, 2023-10-17:
### What kind of device is it?

AIO Coolers

### What is the name of the device?

H150iELITE

### Link to Corsair's Product Page

https://www.corsair.com/us/en/p/cpu-coolers/cw-9060048-ww/icue-h150i-elite-capellix-liquid-cpu-cooler-cw-9060048-ww

### Additional Details

after looking over the supported devices, it seems like the non-elite version appears to work, but using ckb doesn't detect it at all though it does see it in the report

### Generated Report

[ckb-next-report-1697545641.gz](https://github.com/ckb-next/ckb-next/files/12929030/ckb-next-report-1697545641.gz)

gravityfargo, 2024-04-25:
I've added this to the wiki.

The product page says this AIO uses the Commander Core, but the report does not mention it and gives the AIO its own name, `Corsair H150iELITE`, and a product ID of `0c37`. I checked other AIO's reports, and their reports do not mention the AIO by name, only `COMMANDER CORE XT` and/or `COMMANDER CORE`. Which is pretty interesting.

gravityfargo, 2024-05-02:
Closed in favor of #833

closed 2024-05-02 by gravityfargo