ckb-next/ckb-next #1003: Corsair K70 RGB Optical support

paerbac, 2023-10-17:
### What kind of device is it?

Keyboard

### What is the name of the device?

Corsair K70 RGB TKL CHAMPION SERIES Tenkeyless Optical-Mechanical

### Link to Corsair's Product Page

https://www.corsair.com/us/en/p/keyboards/ch-911901a-na/k70-rgb-tkl-champion-series-optical-mechanical-gaming-keyboard-with-pbt-double-shot-pro-keycaps-na-ch-911901a-na

### Additional Details

Device ID is different than the regular K70 RGB TKL Champion

Bus 001 Device 012: ID 1b1c:1bb9 Corsair CORSAIR K70 RGB TKL CHAMPION SERIES Optical Mechanical Gaming Keyboard

### Generated Report

[ckb-next-report-1697583942.gz](https://github.com/ckb-next/ckb-next/files/12968460/ckb-next-report-1697583942.gz)

paerbac, 2023-10-17:
@minneyar @tatokis you guys got the K70 RGB TKL Champion working, could you give me some pointers on how to contribute?

MysteriousRSA, 2024-01-18:
any update on this? I have an OPX and would be willing to test stuff out

Notarin, 2024-02-06:
Welp, looked into this early 2023, made myself forget about it so I may remember later and be pleasantly surprised.
Unfortunately it appears reality had other plans.
I would offer to try, but am not very well acquainted with the code-base and have the pleasure of not having the time to do things I wish to do.
If someone else could entertain this it would enhance my workflow I am attempting to create for myself, if not, well, when I get the time I may do so myself.

gravityfargo, 2024-04-25:
Added to the wiki.

commit c54a82b6 (2025-05-16): Add support for K70 TKL Champ Optic (1bb9)

Resolves: #1003
PR #1157 (2025-05-16): Add support for K70 TKL Champ Optic (1bb9)
Resolves: #1003
closed 2025-05-16 by tatokis