ckb-next/ckb-next #1010: K70 Core Request

CameronReinhart, 2023-11-02:
### What kind of device is it?

Keyboard

### What is the name of the device?

K70 CORE RGB Mechanical Gaming Keyboard — Steel Grey

### Link to Corsair's Product Page

https://www.corsair.com/us/en/p/keyboards/ch-910991e-na/k70-core-rgb-mechanical-gaming-keyboard-steel-grey-ch-910991e-na

### Additional Details

brand new keyboard with incredible value for quality, definitely something that a lot of people are going to end up using in the near future

### Generated Report

[ckb-next-report-1698895935.gz](https://github.com/ckb-next/ckb-next/files/13233939/ckb-next-report-1698895935.gz)

GusGus68, 2023-11-02:
I have the K70 RGB Core UK version. I cannot get it to be recognised, despite following all instructions to update firmware in Windows etc.
Generated Report
[ckb-next-report-1698925997.gz](https://github.com/ckb-next/ckb-next/files/13238452/ckb-next-report-1698925997.gz)
[ckb-next-report-1698925997.gz](https://github.com/ckb-next/ckb-next/files/13238455/ckb-next-report-1698925997.gz)

gravityfargo, 2024-04-25:
Added to the wiki.

nikosszzz, 2024-05-12:
I have the K70 Core RGB, living in Greece so unsure which country model it actually is.
Report: [ckb-next-report-1715515367.gz](https://github.com/ckb-next/ckb-next/files/15286147/ckb-next-report-1715515367.gz)

gravityfargo, 2024-05-12:
> I have the K70 Core RGB, living in Greece so unsure which country model it actually is. Report: [ckb-next-report-1715515367.gz](https://github.com/ckb-next/ckb-next/files/15286147/ckb-next-report-1715515367.gz)

Your device has the same product id as #1019

nikosszzz, 2024-05-12:
> > I have the K70 Core RGB, living in Greece so unsure which country model it actually is. Report: [ckb-next-report-1715515367.gz](https://github.com/ckb-next/ckb-next/files/15286147/ckb-next-report-1715515367.gz)
> 
> Your device has the same product id as #1019

So is it supported? If it is, ckb-next doesn't seem to recognize it.

gravityfargo, 2024-05-13:
@nikosszzz It is not currently supported.

LouKlom, 2024-05-30:
Hi, I'm having the same issue with same keyboard:

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/15503025/ckb-next-dev-detect-report.gz)

Geicomo, 2024-06-27:
Any work done to get this supported, just got one. Thanks for making a program at all!

walter-heymans-corigine, 2024-08-23:
I would like like to see support for this keyboard.

davidbouche, 2024-11-25:
How can I help the project to make this model supported ? Thanks for your project.

t0m5k1, 2025-03-22:
Has there been any progress on this?
We're in March 2025 now.

tatokis, 2025-03-24:
> Has there been any progress on this?

Nope.

You can start by looking at how other similar keyboards were implemented and doing the same https://github.com/ckb-next/ckb-next/commit/86cbf83c024d71bc70c8878525ab31dddee27c15

t0m5k1, 2025-03-24:
> and doing the same 86cbf83

Sorry, Would've done that already if I had the slightest clue what to do.

10j0, 2025-05-10:
🙏

nikosszzz, 2025-05-12:
Will we ever get progress on this keyboard? I currently can't use Linux without having double input behavior which doesn't happen on Windows when I have iCUE installed....

gravityfargo, 2025-08-06:
device now supported
Work was completed in #1157 
commit to main was done in 8b5231f

closed 2025-08-06 by gravityfargo