ckb-next/ckb-next #1022: Support for Corsair K100 AIR ('USB-C' and 'Slipstream Wireless USB Reciver')

fuog, 2023-12-09:
### Enhancement Details

Hello everyone,
I would be interested in support for the **Corsair K100 AIR** (ultra-thin, wireless). This keyboard is still relatively new. 

Thanks for the help 

### Additional Details

With the [latest deb package 0.6.0](https://launchpad.net/ubuntu/+source/ckb-next/0.6.0+dfsg-0.1/+build/26415259) incl. experimental mode I was only able to see the reciver. However, it does not seem to recognize the keyboard itself. The log told me this when i plugged in the reciver (experimental mode):
```log
Dec 09 16:13:25 maulwurf ckb-next-daemon[25614]: [I] Adding device with vendor = 0x1b1c, product = 0x1ba6, path = /dev/bus/usb/001/010, syspath = /sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.1/usb1/1-4
Dec 09 16:13:25 maulwurf ckb-next-daemon[25614]: [I] Connecting CORSAIR SLIPSTREAM WIRELESS USB Receiver at /dev/input/ckb2
Dec 09 16:13:25 maulwurf ckb-next-daemon[25614]: [I] Claiming interfaces. name=CORSAIR SLIPSTREAM WIRELESS USB Receiver, firmware=0505, ep_str= 6
Dec 09 16:13:25 maulwurf ckb-next-daemon[25614]: [I] ckb2: Claiming 6 interfaces
Dec 09 16:13:25 maulwurf ckb-next-daemon[25614]: [I] Starting input thread for /dev/input/ckb2
Dec 09 16:13:25 maulwurf ckb-next-daemon[25614]: [I] Found EP 0x82 at /sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.1/usb1/1-4/1-4:1.0/ep_82
Dec 09 16:13:25 maulwurf ckb-next-daemon[25614]: [I] Endpoint path /sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.1/usb1/1-4/1-4:1.0/ep_82 has wMaxPacketSize 64
Dec 09 16:13:25 maulwurf ckb-next-daemon[25614]: [I] Found EP 0x84 at /sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.1/usb1/1-4/1-4:1.1/ep_84
Dec 09 16:13:25 maulwurf ckb-next-daemon[25614]: [I] Endpoint path /sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.1/usb1/1-4/1-4:1.1/ep_84 has wMaxPacketSize 64
Dec 09 16:13:25 maulwurf ckb-next-daemon[25614]: [I] Found EP 0x83 at /sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.1/usb1/1-4/1-4:1.2/ep_83
Dec 09 16:13:25 maulwurf ckb-next-daemon[25614]: [I] Endpoint path /sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.1/usb1/1-4/1-4:1.2/ep_83 has wMaxPacketSize 64
Dec 09 16:13:25 maulwurf ckb-next-daemon[25614]: [I] Found EP 0x81 at /sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.1/usb1/1-4/1-4:1.3/ep_81
Dec 09 16:13:25 maulwurf ckb-next-daemon[25614]: [I] Endpoint path /sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.1/usb1/1-4/1-4:1.3/ep_81 has wMaxPacketSize 64
Dec 09 16:13:25 maulwurf ckb-next-daemon[25614]: [I] Found EP 0x85 at /sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.1/usb1/1-4/1-4:1.4/ep_85
Dec 09 16:13:25 maulwurf ckb-next-daemon[25614]: [I] Endpoint path /sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.1/usb1/1-4/1-4:1.4/ep_85 has wMaxPacketSize 64
Dec 09 16:13:25 maulwurf ckb-next-daemon[25614]: [I] Found EP 0x86 at /sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.1/usb1/1-4/1-4:1.5/ep_86
Dec 09 16:13:25 maulwurf ckb-next-daemon[25614]: [I] Endpoint path /sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.1/usb1/1-4/1-4:1.5/ep_86 has wMaxPacketSize 64
Dec 09 16:13:25 maulwurf ckb-next-daemon[25614]: [W] bragi_get_property (bragi_common.c:13): Failed to get property 0x15. Device said it's not supported.
Dec 09 16:13:25 maulwurf ckb-next-daemon[25614]: [W] bragi_get_property (bragi_common.c:13): Failed to get property 0x16. Device said it's not supported.
Dec 09 16:13:25 maulwurf ckb-next-daemon[25614]: [W] bragi_get_property (bragi_common.c:13): Failed to get property 0x2. Device said it's not supported.
Dec 09 16:13:25 maulwurf ckb-next-daemon[25614]: [W] bragi_get_property (bragi_common.c:13): Failed to get property 0x44. Device said it's not supported.
Dec 09 16:13:25 maulwurf ckb-next-daemon[25614]: [I] ckb2: Pairing id: 12 b7 9c ae 98 b5 30 fd
Dec 09 16:13:25 maulwurf ckb-next-daemon[25614]: [I] Setup finished for /dev/input/ckb2
```

The USB-C plug was completly ignored by the daemon. _But the Keyboard seams to be intended for cable or wireless usage._

Please find attached the "ckb-next-dev-detect" and further information. [ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/13624498/ckb-next-dev-detect-report.gz)
_(yes, I still have a Corsair Mouse attached, which is irrelevant in this topic)_

I have seen in #1009 that there is already a request for 'K100', but I know that it is not the exact same keyboard and have therefore created a new issue, I hope this is correct.

Unfortunately I can't really help much with the actual coding but if you want me to contribute more information or test a branch, I'm happy to do what I can to support you.

here the regular lsusb naming:
```log
Bus 003 Device 022: ID 1b1c:1bab Corsair CORSAIR K100 RGB AIR WIRELESS Ultra-Thin Mechanical Gaming Keyboard
Bus 003 Device 009: ID 1b1c:1b70 Corsair CORSAIR M55 RGB PRO Gaming Mouse
Bus 001 Device 011: ID 1b1c:1ba6 Corsair CORSAIR SLIPSTREAM WIRELESS USB Receiver
```

gravityfargo, 2024-04-30:
Please use the New Device Request issue template rather than the Enhancement template for requesting support in the future. 

Duplicate of #907

closed 2024-04-30 by gravityfargo

kewur, 2024-06-13:
there is half a support for this in experimental, in cabled mode it works fine but in wireless keymap is wrong. 

I also need to enable experimental for the elite scimitar mouse, is there a way to blacklist this one from ckb-next. I'd rather ckb-next didn't try to do anything with the keyboard but just let the mouse through