ckb-next/ckb-next #1024: Request Support for Virtuoso XT

ThePhaseMaster, 2023-12-20:
### What kind of device is it?

Headphone

### What is the name of the device?

Virtuoso RGB Wireless XT High-Fidelity Gaming Headset — Slate

### Link to Corsair's Product Page

https://www.corsair.com/us/en/p/gaming-headsets/ca-9011188-na/virtuoso-rgb-wireless-xt-high-fidelity-gaming-headset-slate-ca-9011188-na

### Additional Details

Would like to at least control RGB colors, along with toggling off mic light when mic is unmuted.

### Generated Report

[ckb-next-report-1703068510.gz](https://github.com/ckb-next/ckb-next/files/13726225/ckb-next-report-1703068510.gz)

gravityfargo, 2024-05-02:
Could you please unplug the wireless dongle and plug the headset in via USB?

kronflux, 2024-06-18:
[ckb-next-report-1718734027.gz](https://github.com/user-attachments/files/15890759/ckb-next-report-1718734027.gz)
Here is my report with the headset plugged in via USB as well as the Dongle

Literally the only thing keeping me from switching entirely to Linux is the inability to turn off the mic LED when the mic is unmuted, so absolutely anything you might need for this to happen, whether it be reports, logs, testing, even remote system access - I am 100% at your disposal

gravityfargo, 2024-07-12:
Looks like 
0a62 = CORSAIR VIRTUOSO XT USB Gaming Headset
0a64 = CORSAIR VIRTUOSO XT Wireless Gaming Receiver

 I've noticed the Slipstream devices all look the same but their product ids and names are always linked to the device they're sold with. That'll be a fun caveat for whomever works on Slipstream implementation.

kronflux, 2025-01-09:
What other information can we gather in order to implement support for this device?
I have generated two reports. One with the device connected wirelessly via the Slipstream dongle:
[ckb-next-report-1736411264-wireless.gz](https://github.com/user-attachments/files/18358584/ckb-next-report-1736411264-wireless.gz)
and one with the device connected via USB:
[ckb-next-report-1736411352-wired.gz](https://github.com/user-attachments/files/18358585/ckb-next-report-1736411352-wired.gz)

I am happy to test or try anything asked of me, including packet capturing, etc.

kronflux, 2025-01-12:
FYI, OpenLinkHub added this here:
https://github.com/jurkovic-nikola/OpenLinkHub/commit/234a6dd9d674f81878e14f5ddbe83950f6e30e2c

Not sure if you can use that commit to implement into ckb-next, but worth mentioning