ckb-next/ckb-next #1025: NEED HELP

Z3bre-QC, 2023-12-21:
### Describe your situation in detail.

its possible to help me for install on steamos 

i have this error

ckb-next-0.6.0)$ git clone https://github.com/ckb-next/ckb-next.git
cd ckb-next
./quickinstall
Clonage dans 'ckb-next'...
remote: Enumerating objects: 15484, done.
remote: Counting objects: 100% (2952/2952), done.
remote: Compressing objects: 100% (956/956), done.
remote: Total 15484 (delta 2183), reused 2660 (delta 1970), pack-reused 12532
Réception d'objets: 100% (15484/15484), 7.51 Mio | 5.16 Mio/s, fait.
Résolution des deltas: 100% (11425/11425), fait.
./quickinstall: line 51: cmake: command not found
An error occurred,
press enter to exit.

### Additional Details

Add any additional details here.

### Generated Report (if applicable)

Drag your generated report here if it's applicable.

brandon-doornbos, 2023-12-24:
Try OpenRGB, which is in Discover, the app store on your Deck.

Otherwise start searching how to use the AUR on your Deck and use the ckb-next AUR package.

fleischie, 2023-12-24:
@Z3bre-QC 

> ./quickinstall: line 51: cmake: command not found

This means you are probably lacking the build tools. Please refer to your OS' prerequisites (I guess it's Arch, but make sure what the base distribution of your SteamOS is). And install them: https://github.com/ckb-next/ckb-next/wiki/Linux-Installation#build-from-source

closed 2023-12-24 by fleischie