ckb-next/ckb-next #1026: Support for HS80 Max

SinnerK0N, 2023-12-24:
### What kind of device is it?

Headphone

### What is the name of the device?

HS80 MAX WIRELESS Gaming Headset

### Link to Corsair's Product Page

https://www.corsair.com/eu/en/p/gaming-headsets/ca-9011295-eu/hs80-max-wireless-gaming-headset-steel-gray-eu-ca-9011295-eu

### Additional Details

Could be similar to HS80 (#Add Support for) 

### Generated Report

[ckb-next-report-1703453387.gz](https://github.com/ckb-next/ckb-next/files/13762856/ckb-next-report-1703453387.gz)

huntastikus, 2024-03-15:
+1 :D

gravityfargo, 2024-04-30:
This is the EU variant of the HS80, #964 is the German variant. Thank you for searching that ahead of time!

ReDeViLzGz, 2025-01-16:
+1 :) Thanks