ckb-next/ckb-next #1027: Support for the Corsair K95 Platinum SE keyboard

synfy, 2024-01-03:
### What kind of device is it?

Keyboard

### What is the name of the device?

K95 RGB PLATINUM SE Mechanical Gaming Keyboard

### Link to Corsair's Product Page

https://www.corsair.com/us/en/p/keyboards/ch-9127314-na/k95-rgb-platinum-se-mechanical-gaming-keyboard-ch-9127314-na

### Additional Details

Add any additional details here.
ckb-next says there are "no devices detected"

### Generated Report

Drag your generated report here.
[ckb-next-report-1704303823.gz](https://github.com/ckb-next/ckb-next/files/13821912/ckb-next-report-1704303823.gz)

egberts, 2024-01-21:
If you can load up Debug kernel and perform `usbmon` on  your new devices, try following this steps in https://github.com/ckb-next/ckb-next/pull/656#issuecomment-729265011

gravityfargo, 2024-04-30:
@egberts I converted that issue to a wiki page if that helps. I've seen you link to that issue a few times. If you see something that should be changed, please submit an Enhancement issue.
[USB-Capture-Setup](https://github.com/ckb-next/ckb-next/wiki/USB-Capture-Setup)