ckb-next/ckb-next #1028: Layout Portuguese Brazil key "/?" not working

mrmodolo, 2024-01-04:
### Affected Devices

Corsair K60 Pro Mechanical Model RGP0097

### Linux Distribution

Ubuntu 22.04.3 LTS

### Installation Method

git clone https://github.com/ckb-next/ckb-next.git

### ckb-next version

ckb-next 0.6.0-33-gaaf48a0

### What did you expect?

[image]
After install (./quickinstall) the key with "/" stop working. If I uninstall they work again.
[image]

### Did you try anything that may have helped?

I try change key binding for "/" but after that when pressing the key volume increase

### Additional Details

Add any additional details here.
My keyboard layout is Portuguese Brasil.

### Generated Report

Drag your generated report here.
[ckb-next-report-1704380056.gz](https://github.com/ckb-next/ckb-next/files/13831411/ckb-next-report-1704380056.gz)

mrmodolo, 2024-01-04:
Hi!
When the key is working, the keycode follows:

`KeyRelease event, serial 37, synthetic NO, window 0x5c00001,
    root 0x1d4, subw 0x0, time 10341468, (3350,1724), root:(3456,1845),
    state 0x10, keycode 97 (keysym 0x2f, slash), same_screen YES,
    XLookupString gives 1 bytes: (2f) "/"
    XFilterEvent returns: False`

Alexatre, 2024-03-02:
Hi! Thanks for reporting!
The same is happening to me, but my keyboard is the K55 RGB PRO model.

tatokis, 2024-03-02:
Please build ckb-next-daemon with `cmake -DDEBUG_USB_INPUT=1`, run it, and send me the packets that are output when you press and release the key in question while the GUI is running. Make sure brightness is set to 0%, otherwise there will be a lot of spam. And I suggest not to install this build system wide, as it will spam your logs.

Alexatre, 2024-03-04:
Hi!

Thank you very much for your feedback. Below is the new log after the
instructions; I confess that I don't have much experience compiling
packages with CMAKE, so I sent a step-by-step guide of what I did to
generate this new log.

1) Compiled ckb-next-0.6.0 using the command './quickinstall cmake
-DDEBUG_USB_INPUT=1';
2) Enabled the daemon service ('sudo systemctl enable ckb-next-daemon');
3) In the GUI program, selected the value 'Off' for the brightness;
4) Pressed the '?' key a few times to generate the output;
5) Generated the log file clicking in the 'Generated report' button.

If you need me to generate more logs in different ways, I'm available!

Manoel Junior

closed 2024-10-12 by tatokis