ckb-next/ckb-next #1035: Corsair Dark Core RBG Pro

666xan666, 2024-01-28:
### What kind of device is it?

Mouse

### What is the name of the device?

Corsair Dark Core RBG Pro

### Link to Corsair's Product Page

https://www.corsair.com/pl/pl/p/gaming-mouse/ch-9315411-eu/dark-core-rgb-pro-wireless-gaming-mouse-eu-ch-9315411-eu

### Additional Details

Add any additional details here.

### Generated Report

Drag your generated report here.
[ckb-next-report-1706449727.gz](https://github.com/ckb-next/ckb-next/files/14076240/ckb-next-report-1706449727.gz)

NXTler, 2024-03-23:
Duplicate of #698

gravityfargo, 2024-04-23:
@NXTler 
```
Bus 007 Device 004: ID 1b1c:1b80 Corsair CORSAIR DARK CORE RGB PRO Gaming Mouse
```
Specifically *RGB PRO*
This is a different version from the models currently on the wiki, the product ID confirms that.

698 is a pretty messy issue with OP not having had attached logs. 

I'll also inquire on the IRC to brainstorm some way to indicate the issues containing request when a devices is marked as experimental in the wiki.

gensx-x1, 2024-06-23:
i got this mouse few weeks ago and it works on experimental.

Ainstain438, 2025-01-19:
Hi, I got this mouse a while back and I've been having some issues. With the non-experimental daemon (and wireless, not bluetooth mode) the mouse works fine except the special  buttons which write the numbers 1, 2, 3 and 4 and changing the light level on the keyboard doesn't change the brightness of the mouse. It doesn't show up in ckb-next either. This is the log when starting the daemon manually with my K55, Dark Core RGB PRO and slipstream receiver:
```
~ sudo ckb-next-daemon                                                                                                               
ckb-next-daemon 0.6.0-54-g7b36d9e
[I] Root controller ready at /dev/input/ckb0
[I] Connecting Corsair Gaming K55 RGB Keyboard at /dev/input/ckb1
[I] Starting input thread for /dev/input/ckb1
[I] Found EP 0x81 at /sys/devices/pci0000:00/0000:00:07.1/0000:0b:00.3/usb5/5-4/5-4:1.0/ep_81
[I] Received identification packet: 0e 01 00 00 01 01 00 01 08 30 00 03 1c 1b 3d 1b 01 01 01 30 c0 ff 40 00 03 00 00 00 01 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[W] nxp_usb_read (via led_keyboard.c:273): ckb1: Timeout while waiting for response
[W] nxp_usb_read (via led_keyboard.c:273): ckb1: Timeout while waiting for response
[W] nxp_usb_read (via led_keyboard.c:273): ckb1: Timeout while waiting for response
[W] nxp_usb_read (via led_keyboard.c:273): ckb1: Timeout while waiting for response
[W] nxp_usb_read (via led_keyboard.c:273): ckb1: Timeout while waiting for response
[E] _usbrecv (via led_keyboard.c:273): Too many send/recv failures. Dropping.
[W] _start_dev (device.c:202): Unable to load hardware profile
[I] Setup finished for /dev/input/ckb1
```
When I start the daemon in experimental mode (without ckb-next running) this is the log:
```
~ sudo ckb-next-daemon --enable-experimental
ckb-next-daemon 0.6.0-54-g7b36d9e
[I] Root controller ready at /dev/input/ckb0
[I] Connecting CORSAIR SLIPSTREAM WIRELESS USB Receiver at /dev/input/ckb1
[I] Connecting Corsair Gaming K55 RGB Keyboard at /dev/input/ckb2
[I] Starting input thread for /dev/input/ckb2
[I] Found EP 0x81 at /sys/devices/pci0000:00/0000:00:07.1/0000:0b:00.3/usb5/5-4/5-4:1.0/ep_81
[I] Received identification packet: 0e 01 00 00 01 01 00 01 08 30 00 03 1c 1b 3d 1b 01 01 01 30 c0 ff 40 00 03 00 00 00 01 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[I] Starting input thread for /dev/input/ckb1
[I] Found EP 0x82 at /sys/devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-1/1-1:1.0/ep_82
[I] Found EP 0x84 at /sys/devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-1/1-1:1.1/ep_84
[I] Found EP 0x83 at /sys/devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-1/1-1:1.2/ep_83
[I] Found EP 0x81 at /sys/devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-1/1-1:1.3/ep_81
[I] Found EP 0x85 at /sys/devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-1/1-1:1.4/ep_85
[I] Found EP 0x86 at /sys/devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-1/1-1:1.5/ep_86
[E] process_input_urb (keymap.c:837): Unknown bragi data received in input thread 00 from endpoint 85
[I] ckb1: bragi dongle hotplug thread started
[I] ckb1: bragi dongle hotplug thread finished
[I] ckb1: bragi dongle hotplug thread started
[I] Found new bragi subdevice 1
[W] bragi_get_property (bragi_common.c:13): Failed to get property 0x15. Device said it's not supported.
[W] bragi_get_property (bragi_common.c:13): Failed to get property 0x16. Device said it's not supported.
[I] ckb3: Subkb vendor: 0x1b1c, product: 0x1b80
[I] ckb1: bragi dongle hotplug thread finished
[W] bragi_get_property (bragi_common.c:13): Failed to get property 0x2. Device said it's not supported.
[W] bragi_get_property (bragi_common.c:13): Failed to get property 0x44. Device said it's not supported.
[I] ckb1: Pairing id: f8 d6 75 2a 33 7e 4b fe 
[I] Setup finished for /dev/input/ckb1
[W] bragi_process_notification (bragi_notification.c:36): ckb1: Unknown bragi notification for dev 3
[W] bragi_process_notification (bragi_notification.c:36): ckb1: Unknown bragi notification for dev 3
[W] bragi_get_property (bragi_common.c:13): Failed to get property 0x15. Device said it's not supported.
[W] bragi_get_property (bragi_common.c:13): Failed to get property 0x16. Device said it's not supported.
[I] ckb3: Pairing id: f8 d6 75 2a 33 7e 4b fe 
[I] Setup finished for /dev/input/ckb3
[E] hid_mouse_translate (keymap.c:1004): Invalid length 4
[E] hid_mouse_translate (keymap.c:1004): Invalid length 4
[E] hid_mouse_translate (keymap.c:1004): Invalid length 4
[E] hid_mouse_translate (keymap.c:1004): Invalid length 4
[E] hid_mouse_translate (keymap.c:1004): Invalid length 4
[E] hid_mouse_translate (keymap.c:1004): Invalid length 4
[E] hid_mouse_translate (keymap.c:1004): Invalid length 4
[E] hid_mouse_translate (keymap.c:1004): Invalid length 4
[E] hid_mouse_translate (keymap.c:1004): Invalid length 4
[E] hid_mouse_translate (keymap.c:1004): Invalid length 4
[E] hid_mouse_translate (keymap.c:1004): Invalid length 4
[E] hid_mouse_translate (keymap.c:1004): Invalid length 4
[E] hid_mouse_translate (keymap.c:1004): Invalid length 4
[E] hid_mouse_translate (keymap.c:1004): Invalid length 4
[E] hid_mouse_translate (keymap.c:1004): Invalid length 4
[E] hid_mouse_translate (keymap.c:1004): Invalid length 4
[... 51 lines cut ...]
[I] Stopping indicator thread for ckb1 (4)
[I] Stopping input thread for /dev/input/ckb1
[... cut ...] e of the screen. This makes the mouse somewhat impractical to use lol

## Wired (USB-C)

In wired USB-C mode the behaviour is similar when run with non experimental daemon (doesn't show up in ckb-next) and the cursor doesn't jump with the experimental daemon. (with ckb-next active).

## Generated report

Here is my report generated with ckb-next running and the experimental daemon. 

[ckb-next-report-1737309327.gz](https://github.com/user-attachments/files/18469797/ckb-next-report-1737309327.gz)