ckb-next/ckb-next #1037: K100 RGB Optical wheel action binding

Fail-gh, 2024-02-13:
### Enhancement Details

Add the binding option to the wheel, now is useless

kimjongbing, 2024-07-23:
I want to add onto this that the wheel right now (not the button inside the wheel) if slightly touched or spun will cause erratic behavior like the triggering of many keys on the keyboard and page direction (scrolling up/down) and navigating to other pages (back/forward).

Azyrion, 2025-03-18:
the wheel on the keyboard is definitely useless and also causes a lot of trouble. No idea why but it posted some nonsense in a discord group that nearly kicked me out! So the software is doin something with it.....can you at least unbind it?

mmtmn, 2025-06-29:
KEY_FASTFORWARD (208) → clockwise

KEY_REWIND (168) → counter-clockwise

working on a PR for this

mmtmn, 2025-07-13:
@Azyrion yeah, I fixed it, its in this pull request: https://github.com/ckb-next/ckb-next/pull/1172

no idea why it hasn't been approved or reviewed by @tatokis or @Ravenslofty

mmtmn, 2025-08-05:
yeah pretty sus open source repository I guess, i've seen other suggest valid PRs and then get pretty badly and unfairly treated, I'm not trusting this anymore, so i'll do what other did and just close it and never use and recommend against using it

KABoissonneault, 2026-01-20:
It's a bit unfortunate what happened to the PR. This issue is actually quite bad, accidentally hitting that wheel will send random inputs all over. I don't really need support to bind it to anything relevant, I'd like if I could just disable it.

Ravenslofty, 2026-01-20:
Glad to hear it.

MadMartian, 2026-06-07:
Mapping it to volume control is better than nothing, currently it crashes [keyd](https://github.com/rvaiya/keyd) with a panic, which I'm using to manage input from a different macropad.  I suppose installing the latest is a better option that dumping a wad of hot plastic glue on the knob so I never accidentally bump it.