ckb-next/ckb-next #1045: Keyboard lights no longer sleeping on idle

Retrograde-i486, 2024-03-05:
### Affected Devices

Corsair K95 Platinum

### Linux Distribution

Garuda Linux

### Installation Method

AUR

### ckb-next version

ckb-next 0.6.0

### What did you expect?

Keyboard no longers turns of illuminated keys after a set period of time. Setting 'Turn lgiths off when idle' doesn't seem to work anymore but it did previously.

### Did you try anything that may have helped?

I've tried a range of idle timeout times such as 1 minute, 5 minute and 10 minutes. I've tried several different profiles I have set up and none of them turn off when the idle timer set in the settings is reached.

### Additional Details

Add any additional details here.

### Generated Report

[ckb-next-report-1709602175.gz](https://github.com/ckb-next/ckb-next/files/14489149/ckb-next-report-1709602175.gz)

thetundraterror, 2024-04-08:
Yeah, I'm having pretty similar issues in that I can't even enable the _Turn off lights when idle_.

thetundraterror, 2024-04-24:
Seems to be an issue when using Wayland as swapping to X11 allows for the setting to work. Not sure if it's something that can be "fixed" in Wayland due to how the compositor functions.

Retrograde-i486, 2024-04-24:
> Seems to be an issue when using Wayland as swapping to X11 allows for the setting to work. Not sure if it's something that can be "fixed" in Wayland due to how the compositor functions.

Yeah that makes sense given it started right after switching to KDE plasma 6 which comes with Wayland by default.