ckb-next/ckb-next #1046: Corsair Voyager a1600 Laptop Keyboard Support Request

james309, 2024-03-05:
### What kind of device is it?

Keyboard

### What is the name of the device?

CORSAIR VOYAGER a1600 Gaming Laptop

### Link to Corsair's Product Page

https://www.corsair.com/us/en/p/gaming-computers/cn-9000003-na/corsair-voyager-a1600-gaming-laptop-amd-advantage-edition-ryzen-r7-radeon-6800m-16gb-ddr5-1tb-m-2-16in-240hz-display-win11-home-cn-9000003-na

### Additional Details

Weirdly after installing and rebooting it reset the keyboard RGB to factory as rainbow transition from what i had set in windows 11. Makes me think it should work, but won't show up in the device section. I did trim out a lot of other Corsair devices in the dev report (slipstream module, stream deck, & camera all built in to the laptop).

### Generated Report

[ckb-next-report-1709663692-trimmed.gz](https://github.com/ckb-next/ckb-next/files/14500511/ckb-next-report-1709663692-trimmed.gz)

linuxazo, 2024-10-18:
I too could help test this keyboard.  I use Debian.  Thank you so much in advance.

cooler1972, 2025-05-02:
Hello, I hope this helps. Linux Mint 22.1 Kernel 6.11.0-24, Manufacturer code of laptop: CN-9000003-UK 

Bus 001 Device 008: ID 1b1c:1bd0 Corsair CORSAIR ULP Laptop Keyboard 
usbhid-dump -m 1b1c:1bd0:

001:008:003:DESCRIPTOR         1746156471.250408
 05 01 09 02 A1 01 85 05 09 01 A1 00 05 09 19 01
 29 20 15 00 25 01 95 20 75 01 81 02 05 01 09 30
 09 31 16 01 00 26 FF 7F 75 10 95 02 81 06 09 38
 15 81 25 7F 75 08 95 01 81 06 C0 C0

001:008:002:DESCRIPTOR         1746156471.251285
 06 42 FF 09 02 A1 01 15 00 26 FF 00 95 40 75 08
 09 02 81 02 C0

001:008:001:DESCRIPTOR         1746156471.251757
 06 42 FF 09 01 A1 01 15 00 26 FF 00 95 40 75 08
 09 01 81 02 95 40 09 01 91 02 C0

001:008:000:DESCRIPTOR         1746156471.252247
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 29 97 15 00 25 01
 75 01 95 98 81 02 05 08 19 01 29 03 25 01 75 01
 95 03 91 02 75 05 95 01 91 01 C0 05 0C 09 01 A1
 01 85 02 05 0C 19 00 2A FF 0F 15 00 26 FF 0F 75
 10 95 02 81 00 C0 05 01 09 80 A1 01 85 06 19 81
 29 83 15 00 25 01 95 03 75 01 81 02 95 01 75 05
 81 01 C0 05 01 09 0C A1 01 85 04 15 00 25 01 09
 C6 95 01 75 01 81 06 75 07 81 03 09 C7 75 01 91
 02 75 07 91 03 C0

Bus 001 Device 006: ID 1b1c:0d02 Corsair CORSAIR Stream Deck Bar 
usbhid-dump -m 1b1c:0d02:

001:006:000:DESCRIPTOR         1746155411.876055
 05 0C 09 01 A1 01 09 01 05 09 19 01 29 FF 15 00
 26 FF 00 75 08 96 3F 00 85 01 81 02 0A 00 FF 15
 00 26 FF 00 75 08 96 FF 03 85 02 91 02 0A 00 FF
 15 00 26 FF 00 75 08 96 1F 00 85 03 B1 04 0A 00
 FF 15 00 26 FF 00 75 08 96 1F 00 85 04 B1 04 0A
 00 FF 15 00 26 FF 00 75 08 96 1F 00 85 05 B1 04
 0A 00 FF 15 00 26 FF 00 75 08 96 1F 00 85 06 B1
 04 0A 00 FF 15 00 26 FF 00 75 08 96 1F 00 85 07
 B1 04 0A 00 FF 15 00 26 FF 00 75 08 96 1F 00 85
 08 B1 04 0A 00 FF 15 00 26 FF 00 75 08 96 1F 00
 85 09 B1 04 0A 00 FF 15 00 26 FF 00 75 08 96 1F
 00 85 0A B1 04 0A 00 FF 15 00 26 FF 00 75 08 96
 1F 00 85 0B B1 04 0A 00 FF 15 00 26 FF 00 75 08
 96 1F 00 85 0C B1 04 0A 00 FF 15 00 26 FF 00 75
 08 96 1F 00 85 0D B1 04 0A 00 FF 15 00 26 FF 00
 75 08 96 1F 00 85 0E B1 04 0A 00 FF 15 00 26 FF
 00 75 08 96 1F 00 85 0F B1 04 0A 00 FF 15 00 26
 FF 00 75 08 96 1F 00 85 10 B1 04 0A 00 FF 15 00
 26 FF 00 75 08 96 1F 00 85 11 B1 04 0A 00 FF 15
 00 26 FF 00 75 08 96 1F 00 85 12 B1 04 0A 00 FF
 15 00 26 FF 00 75 08 96 1F 00 85 13 B1 04 C0

Furthermore a user posted a probe of this laptop : https://linux-hardware.org/?probe=f7d5ff0208

cooler1972, 2025-05-19:
[image]

I can't say I know what I'm doing, but it does show up in ckb-next if I try hard throwing stuff in hard enough... ofc the keyboard does not register inputs, neither does changing lighting via software.