ckb-next/ckb-next #1049: Corsair CORSAIR K55 CORE RGB Gaming Keyboard

Lila-Kuh, 2024-04-04:
### What kind of device is it?

Keyboard

### What is the name of the device?

Corsair CORSAIR K55 CORE RGB Gaming Keyboard

### Additional Details

ckb-next-dev-detect - Thu Apr  4 15:16:32 CEST 2024

```
Kernel: Linux MSIMjO1902 6.6.24-1-MANJARO #1 SMP PREEMPT_DYNAMIC Thu Apr  4 04:33:21 UTC 2024 x86_64 GNU/Linux

OS Version:
LSB Version:	n/a
Distributor ID:	ManjaroLinux
Description:	Manjaro Linux
Release:	23.1.4
Codename:	Vulcan

USB Devices:

Bus 003 Device 002: ID 1b1c:1bfe Corsair CORSAIR K55 CORE RGB Gaming Keyboard
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1b1c Corsair
  idProduct          0x1bfe CORSAIR K55 CORE RGB Gaming Keyboard
  bcdDevice           14.04
  iManufacturer           1 Corsair
  iProduct                2 CORSAIR K55 CORE RGB Gaming Keyboard
  iSerial                 3 AD000000249E420062D795102BB00000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x005b
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     156
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      27
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      21
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)

Dmesg on current boot:
Apr 04 15:13:28.648333 archlinux kernel: usb 3-4: New USB device found, idVendor=1b1c, idProduct=1bfe, bcdDevice=14.04
Apr 04 15:13:28.648733 archlinux kernel: usb 3-4: Product: CORSAIR K55 CORE RGB Gaming Keyboard
Apr 04 15:13:28.648869 archlinux kernel: usb 3-4: Manufacturer: Corsair
Apr 04 15:13:28.651635 archlinux kernel: input: Corsair CORSAIR K55 CORE RGB Gaming Keyboard as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/0003:1B1C:1BFE.0002/input/input5
Apr 04 15:13:28.711666 archlinux kernel: hid-generic 0003:1B1C:1BFE.0002: input,hidraw1: USB HID v1.10 Keyboard [Corsair CORSAIR K55 CORE RGB Gaming Keyboard] on usb-0000:00:14.0-4/input0
Apr 04 15:13:28.711933 archlinux kernel: hid-generic 0003:1B1C:1BFE.0003: hiddev96,hidraw2: USB HID v1.10 Device [Corsair CORSAIR K55 CORE RGB Gaming Keyboard] on usb-0000:00:14.0-4/input1
Apr 04 15:13:28.712101 archlinux kernel: hid-generic 0003:1B1C:1BFE.0004: hiddev97,hidraw3: USB HID v1.10 Device [Corsair CORSAIR K55 CORE RGB Gaming Keyboard] on usb-0000:00:14.0-4/input2
Apr 04 15:13:28.648333 archlinux kernel: usb 3-4: New USB device found, idVendor=1b1c, idProduct=1bfe, bcdDevice=14.04
Apr 04 15:13:28.648733 archlinux kernel: usb 3-4: Product: CORSAIR K55 CORE RGB Gaming Keyboard
Apr 04 15:13:28.648869 archlinux kernel: usb 3-4: Manufacturer: Corsair
Apr 04 15:13:28.651635 archlinux kernel: input: Corsair CORSAIR K55 CORE RGB Gaming Keyboard as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/0003:1B1C:1BFE.0002/input/input5
Apr 04 15:13:28.711666 archlinux kernel: hid-generic 0003:1B1C:1BFE.0002: input,hidraw1: USB HID v1.10 Keyboard [Corsair CORSAIR K55 CORE RGB Gaming Keyboard] on usb-0000:00:14.0-4/input0
Apr 04 15:13:28.711933 archlinux kernel: hid-generic 0003:1B1C:1BFE.0003: hiddev96,hidraw2: USB HID v1.10 Device [Corsair CORSAIR K55 CORE RGB Gaming Keyboard] on usb-0000:00:14.0-4/input1
Apr 04 15:13:28.712101 archlinux kernel: hid-generic 0003:1B1C:1BFE.0004: hiddev97,hidraw3: USB HID v1.10 Device [Corsair CORSAIR K55 CORE RGB Gaming Keyboard] on usb-0000:00:14.0-4/input2
Apr 04 15:13:51.091625 MSIMjO1902 kernel: warning: `ckb-next.real' uses wireless extensions which will stop working for Wi-Fi 7 hardware; use nl80211

Dmesg on last boot:
Apr 04 12:24:02.511649 MSIMjO1902 kernel: warning: `ckb-next.real' uses wireless extensions which will stop working for Wi-Fi 7 hardware; use nl80211

Daemon on current boot:
-- No entries --

Daemon on last boot:
-- No entries --

Dev nodes:
/dev/input/ckb0
/dev/input/ckb0/connected
/dev/input/ckb0/version
0.6.0-33-gaaf48a03
/dev/input/ckb0/pid
900

Kernel command line:
BOOT_IMAGE=/boot/vmlinuz-6.6-x86_64 root=UUID=da955b06-ba26-41ac-a205-514346548560 rw udev.log_priority=3

GUI ldd:

Environment:
XDG_SESSION_DESKTOP=xfce
QT_QPA_PLATFORMTHEME=qt6ct
XDG_SESSION_TYPE=x11
XDG_CURRENT_DESKTOP=XFCE
DISPLAY=:0.0
QT_AUTO_SCREEN_SCALE_FACTOR=0
```

### Generated Report

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/14872032/ckb-next-dev-detect-report.gz)

vimalKeshu, 2024-08-13:
Hi @gravityfargo waiting for this driver; any plan date for release ? if it's needed, I am ready to help/contribute; i am new to linux driver but still if someone guide, i can contribute. thank you.

TheKamboy, 2024-08-31:
i am also hoping this will be added

TheKamboy, 2024-08-31:
@gravityfargo

imnotamessiah, 2024-08-31:
hey there, yeah me too please

Inmapi, 2026-09-21:
I got one ( K55 RGB Core - id: 1bfe) I am available for testing on Linux Mint.