ckb-next/ckb-next #1052: Daemon fails to start after enabling experminatal support on boot, and refuses to start even after removing the lines

Richterman, 2024-04-15:
### Describe your situation in detail.

I pasted the lines to get it to start at boot so my keyboard would change color at boot instead of log in, The daemon decided to no longer run after reboot, I removed the lines, rebooted still won't start with systemctl. I uninstalled it from fedora and then built from source, still didn't work

I'm at a loss of what to do, journalctl won't show me anything

Fedora 40

### Additional Details

Add any additional details here.
Apr 14 22:32:17 fedora systemd[1]: ckb-next-daemon.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ The unit ckb-next-daemon.service has entered the 'failed' state with result 'exit-code'.
Apr 14 22:32:17 fedora systemd[1]: Failed to start ckb-next-daemon.service - Corsair Keyboards and Mice Daemo>
░░ Subject: A start job for unit ckb-next-daemon.service has failed
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ A start job for unit ckb-next-daemon.service has finished with a failure.
░░ 
░░ The job identifier is 10009 and the job result is failed.
lines 1110-1136/1136 (END)

### Generated Report (if applicable)

Drag your generated report here if it's applicable.
[ckb-next-report-1713152158.gz](https://github.com/ckb-next/ckb-next/files/14973668/ckb-next-report-1713152158.gz)
[ckb-next-report-1713152158.gz](https://github.com/ckb-next/ckb-next/files/14973669/ckb-next-report-1713152158.gz)

tatokis, 2024-04-15:
Apologies for this. I suspect it's because Fedora changed the daemon path. Clearing out all the lines doesn't actually delete the override. Have you tried `sudo systemctl revert ckb-next-daemon` ?

As for the guide, someone in another ticket some time ago suggested a better solution to enable experimental support, but I never got around to actually writing how to do it in the wiki.

Richterman, 2024-04-15:
That worked, thank you

I didn't even know that was a systemd option. I usually just stop then start it. 

Thank you