ckb-next/ckb-next #1053: Corsair K70 CORE RGB

dacron74, 2024-04-21:
### What kind of device is it?

Keyboard

### What is the name of the device?

Corsair K70 CORE RGB

### Link to Corsair's Product Page

https://www.corsair.com/de/de/p/keyboards/ch-910981e-de/k70-core-rgb-mechanical-gaming-keyboard-with-palm-rest-black-de-ch-910981e-de#tab-downloads

### Additional Details

Add any additional details here.

### Generated Report

Drag your generated report here.
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/15051231/ckb-next-dev-detect-report.gz)

arizonagroovejet, 2024-04-23:
Here's a report for the same device on Fedora Workstation.
[ckb-next-report-1713905598.gz](https://github.com/ckb-next/ckb-next/files/15082811/ckb-next-report-1713905598.gz)

gravityfargo, 2024-04-30:
Duplicate of #1019

gravityfargo, 2024-04-30:
Note: This German variant of the K70 CORE RGB shares the same product as the UK variant. This has been annotated in the wiki.

closed 2024-04-30 by gravityfargo