ckb-next/ckb-next #1054: Vengeance M95 configuration mostly broken.

OscarVY, 2024-04-23:
### Affected Devices

Vengeance M95

### Linux Distribution

Linux Mint 21.3 Cinnamon 

### Installation Method

mintinstall - Software manager

### ckb-next version

ckb-next 0.6.0

### What did you expect?

- The GUI's DPI sliders do nothing to change mouse sensitivity.

- Mouse sensitivity changes randomly on restart or when the mouse is un- and replugged. High variance, feels like between 100 and 3000 dpi, tending towards the extremes.

- y-axis sensitivity is consistently higher than x-axis sensitivity, giving the mouse a 'jittery' feel. Independent X/Y states is unchecked.

- DPI sliders max out at 820 DPI. Manual input of the numbers gets capped at 820 as well. Again, these sliders do nothing anyway.

- Changes to the poll rate seem to have no impact and do not get saved.

- The forwards and back functionality of button 7 & 8 breaks once ckb-next starts, even when the active mode is reset to default. Rebinding these buttons with ckb-next's equivalents yields no results.

- Saving to hardware and restarting without ckb-next gave random and unhelpful results. As an example, once all keys (including side buttons and mousewheel click, MW up and MW down) got rebound to output left click.

### Did you try anything that may have helped?

Multiple reboots, fiddling with various GUI options, unplugging the mouse, reinstalling ckb, etc

### Additional Details

Mouse sensitivity is perfectly fine while ckb-next and it's deamon aren't running.
Firmware version is: 2.24

### Generated Report

[ckb-next-report-1713906123.gz](https://github.com/ckb-next/ckb-next/files/15083191/ckb-next-report-1713906123.gz)

LiamM32, 2024-05-19:
I am getting the same issue with trying to bind the forward and back buttons. They were already set prior to launching ckb-next, but now they aren't working, even after binding them in this program.

I am on Manjaro Linux.
KDE Plasma version 6.0.4
KDE Frameworks version 6.1.0
X11
Linux Kernel version 6.6.30-MANJARO

LiamM32, 2024-05-19:
It turns out that it's worse than I thought. After using the "Save to hardware" option and restarting my computer in either Windows 10 or Linux, the mouse tracker has just stopped working entirely. It only starts working again if I start `ckb-next-daemon` in Linux.

In essence, the mouse has become dependent on CKB-next for the tracker to work.