ckb-next/ckb-next #1058: Cursor keys don't work after reboot

aaronw2, 2024-05-02:
### Affected Devices

K95 Pro RGB

### Linux Distribution

OpenSUSE 15.5

### Installation Method

RPM

### ckb-next version

ckb-next 0.6.0-33-gaaf48a03

### What did you expect?

After rebooting, some of the cursor keys do not work. Unplugging and replugging the keyboard into the USB port fixes this. It doesn't happen every time, but very frequently when rebooting.

### Additional Details

Add any additional details here.

### Generated Report

Drag your generated report here.
[ckb-next-report-1714619189.gz](https://github.com/ckb-next/ckb-next/files/15183365/ckb-next-report-1714619189.gz)

gravityfargo, 2024-05-02:
Seems this is the offending error:
Startup
```
Apr 30 16:40:03.207790 flash ckb-next-daemon[5444]: [I] Stopping input thread for /dev/input/ckb1
Apr 30 16:40:03.231449 flash ckb-next-daemon[5444]: [I] Disconnecting /dev/input/ckb1
Apr 30 16:40:03.231669 flash ckb-next-daemon[5444]: [I] Caught internal signal SIGUSR2
Apr 30 16:40:03.231669 flash ckb-next-daemon[5444]: [I] Stopping indicator thread for ckb1 (4)
Apr 30 16:40:03.453504 flash ckb-next-daemon[5444]: [W] usbunclaim (usb_linux.c:261): Your Corsair K95 RGB Gaming Keyboard is being uncleanly removed to speed up shutdown times.
Apr 30 16:40:03.453504 flash ckb-next-daemon[5444]: [W] usbunclaim (usb_linux.c:262): If you still need the device, you will have to restart ckb-next-daemon.
Apr 30 16:40:03.453761 flash ckb-next-daemon[5444]: [I] Removed device path /dev/input/ckb1
Apr 30 16:40:04.902071 flash ckb-next-daemon[5444]: [I] Connecting Corsair K95 RGB Gaming Keyboard at /dev/input/ckb1
```
Shutdown
```
Apr 30 16:22:01.939202 flash ckb-next-daemon[5470]: [I] Caught signal 15
Apr 30 16:22:01.938660 flash systemd[1]: Stopping Corsair Keyboards and Mice Daemon...
Apr 30 16:22:02.086660 flash ckb-next-daemon[5470]: [I] Disconnecting /dev/input/ckb1
Apr 30 16:22:02.088801 flash ckb-next-daemon[5470]: [I] Caught internal signal SIGUSR2
Apr 30 16:22:02.088801 flash ckb-next-daemon[5470]: [I] Stopping indicator thread for ckb1 (4)
Apr 30 16:22:02.611365 flash ckb-next-daemon[5470]: [I] Stopping input thread for /dev/input/ckb1
Apr 30 16:22:02.611365 flash ckb-next-daemon[5470]: [W] usbunclaim (usb_linux.c:261): Your Corsair K95 RGB Gaming Keyboard is being uncleanly removed to speed up shutdown times.
Apr 30 16:22:02.611365 flash ckb-next-daemon[5470]: [W] usbunclaim (usb_linux.c:262): If you still need the device, you will have to restart ckb-next-daemon.
```

aaronw2, 2024-05-04:
The shutdown time on my desktop can be rather lengthy, given the many mount points and the fact that a VM also needs to be cleanly shut down.

gravityfargo, 2024-05-13:
A temporary workaround would be to add a timeout to the service file, but that will probably have unintended consequences. 
[systemd service docs](https://www.freedesktop.org/software/systemd/man/latest/systemd.kill.html)

```
sudo systemctl edit --full ckb-next-daemon.service     
```

```
# Copyright 2017-2018 ckb-next Development Team <ckb-next@googlegroups.com>
# Distributed under the terms of the GNU General Public License v2

[Unit]
Description=Corsair Keyboards and Mice Daemon

[Service]
ExecStart=/usr/bin/ckb-next-daemon
# Restart=on-failure
TimeoutSec=10s

[Install]
WantedBy=multi-user.target
```

You could also just disable auto-start all together . Then manually start the daemon after login, or use an auto-start script, but you'll be asked for admin permissions on login. 

```
sudo systemctl disable ckb-next-daemon.service 
```

[Archwiki on auto-starting via desktop environment](https://wiki.archlinux.org/title/Autostarting#On_desktop_environment_startup)