ckb-next/ckb-next #1060: K65 PLUS Wireless

UngraguGaming, 2024-05-14:
### What kind of device is it?

Keyboard

### What is the name of the device?

K65 PLUS WIRELESS 75% RGB Mechanical Gaming Keyboard

### Link to Corsair's Product Page

https://www.corsair.com/us/en/p/keyboards/ch-91d401l-na/k65-plus-wireless-75-percent-rgb-mechanical-gaming-keyboard-ch-91d401l-na

### Additional Details

 Report lists it as being detected as follows:

``` 
Bus 001 Device 004: ID 1b1c:2b10 Corsair CORSAIR K65 PLUS WIRELESS Keyboard
```
There's also something I noticed about it's "bcountrycode" (I do not know what this is) being not supported.

I'm not sure if this is important or not, but I'll include this anyways. The keyboard has limited onboard memory in wireless mode, and as a result, it tends to default to preset lighting effects when not plugged in.

### Generated Report

[ckb-next-report-1715650343.gz](https://github.com/ckb-next/ckb-next/files/15301497/ckb-next-report-1715650343.gz)

MSalman6, 2024-08-31:
Hi, i've just gotten the keyboard and it is very buggy (Some keys don't work and does auto inputs) on onboard memory, I've tried on Windows, Ubuntu and even on Mobile.

It only works properly with iCUE installed on Windows and is completely useless everywhere else.

Is ckb-next only for controlling the RBG on the keyboard or it will help in this case?

MSalman6, 2024-08-31:
Hi, I've just received the keyboard, and it is very buggy (some keys don't work, and it makes auto inputs) when using the onboard memory. I've tried it on Windows, Ubuntu, and even on mobile.

It only works properly with iCUE installed on Windows and is completely useless everywhere else.

Is ckb-next only for controlling the RGB on the keyboard, or will it help in this case?

MSalman6, 2024-11-22:
> > Hi, I've just received the keyboard, and it is very buggy (some keys don't work, and it makes auto inputs) when using the onboard memory. I've tried it on Windows, Ubuntu, and even on mobile.
> > 
> > It only works properly with iCUE installed on Windows and is completely useless everywhere else.
> > 
> > Is ckb-next only for controlling the RGB on the keyboard, or will it help in this case?
> 
> I have run into the exact same thing. Regardless of whether I'm on Windows, Linux, or even MacOS, whenever I press period or the right arrow, I get a consistent string of characters (none of which include the original character) unless I have iCue enabled. I'm hoping that this will allow me to fully use my keyboard since this has been really annoying me
> 
> maybe the string could be some sort of protocol meant to communicate with iCue, but I'm not sure

Force updating the firware from the iCue software for both the keyboard and receiver did fix the issue with the period key but now I have issue of frequent disconnections over the dongle although it sits very close.

You can also try resetting the keyboard by following relevant guide on their site

MSalman6, 2024-11-22:
> I also have very frequent disconnects. To the point I have to keep the keyboard on wired mode, otherwise it's way too disruptive. Happens only on Linux, under Windows it seems fine. The only way to get it working again after it disconnects is by unplugging and plugging back the dongle.

Yes, I have to unplug and plug the dongle again to get it working again also.

Though after the very recent firmware update it hasn't happened have you tried it?

earthly-alien, 2024-11-22:
I haven't tried yet (I just keep the kb plugged all the time now), but I'll give it a shot. Thanks for the heads up.

wedgevic, 2024-11-22:
> Force updating the firware from the iCue software for both the keyboard and receiver did fix the issue with the period key but now I have issue of frequent disconnections over the dongle although it [...]

I use the keyboard wired since I lost my adapter a while back, but unfortunately resetting it and re-flashing the firmware did not fix it for me. I will be opening a Corsair support ticket and update if they find any fix

earthly-alien, 2024-11-24:
> > I also have very frequent disconnects. To the point I have to keep the keyboard on wired mode, otherwise it's way too disruptive. Happens only on Linux, under Windows it seems fine. The only way to get it working again after it disconnects is by unplugging and plugging back the dongle.
> 
> Yes, I also have to unplug and plug the dongle to get it working again.
> 
> Though after the very recent firmware update it hasn't happened.

Unfortunately, even after updating firmware my keyboard still loses connection occasionally. And now for some reason my keyboard layout isn't recognized anymore (I can't get all the keys to align to their respective values, but maybe that's caused by Fedora upgrade).

AlexanderLB98, 2025-01-29:
Hello, I am also having issues with this keyboard. I have to use it almost always wired because with the wireless usb it loses connection from time to time and I have to reconnect it. I cant make it work with bluetooth either, nor in Arch and PopOS. Works flawlessly on Windows.

Absolemo, 2025-05-18:
The keyboard runs fine under linux. So far i haven't noticed any crashes. I would also like to be able to set the rgbs but no devices are found. The onboard memory from corsair is completely obsolete. Only works in wireless mode. I just want to change the color of my keyboard to a nice looking orange, plain and simple.

Dmytro-Astafurov, 2025-09-05:
> I just came across this awesome open-source Linux project that supports almost every Corsair device including K65 Plus.
> 
> Initial thoughts - everything works, it fixed buggy and auto inputs, RGB Lighting, option to create and save profiles, knob control etc.
> 
> Everyone should give it a try - https://github.com/jurkovic-nikola/OpenLinkHub

Thank you for this information!

wedgevic, 2025-09-10:
> I just came across this awesome open-source Linux project that supports almost every Corsair device including K65 Plus.
>
> Initial thoughts - everything works, it fixed buggy and auto inputs, RGB [...]

Thank you, this finally managed to fix it up for me, and I can finally use my keyboard as intended