ckb-next/ckb-next #1062: Corsair K70 PRO Mechanical Gaming Keyboard

InternalAardvark, 2024-05-23:
### What kind of device is it?

Keyboard

### What is the name of the device?

CORSAIR K70 RGB PRO Mechanical Gaming Keyboard

### Link to Corsair's Product Page

https://www.corsair.com/us/en/p/keyboards/ch-910941a-na/k70-pro-rgb-optical-mechanical-gaming-keyboard-with-pbt-double-shot-pro-keycaps-ch-910941a-na

### Additional Details

Add any additional details here.
The keyboard i have is showing as 1bc4 under lsusb witch is not in the supported devies

### Generated Report

Drag your generated report here.
[ckb-next-report-1716447488.gz](https://github.com/ckb-next/ckb-next/files/15412933/ckb-next-report-1716447488.gz)

meduzapat, 2025-03-25:
@InternalAardvark  I added code to support this device, I tested it and works fine.

DarkeFae, 2025-04-06:
> [@InternalAardvark](https://github.com/InternalAardvark) I added code to support this device, I tested it and works fine. You will be able to use it as soon as is merged.

I have a 1bc4, sadly your code is for 1bd4 and does not work with the 1bc4 from my testing

RuralRob, 2025-04-27:
I also have a 1bc4. Keyboard works fine but shows up as "no devices connected". Here is my report:

[ckb-next-report-1745763360.gz](https://github.com/user-attachments/files/19930230/ckb-next-report-1745763360.gz)

michaeldejong-ca, 2025-08-22:
Hi there, I just recently got a K70-PRO to replace an older Corsair keyboard, it was quite old.
However CKB Next does not recognize the new device like it did the earlier one.
I did some homework about the 'bios mode' but that didn't help. If someone can help me that would be great. Thanks :-)

michaeldejong-ca, 2025-09-24:
That's great meduzapat, when should those changes take place?