ckb-next/ckb-next #1065: Enabling experimental mode on Fedora - command not found

jacob99200, 2024-05-26:
### Describe your situation in detail.

I'm trying to enable experimental mode on Fedora but it seems the ckb-next-daemon command cant be found

how would I go about fixing this ? or is there an alternative to test out experimental mode?

the ckb-next-daemon.service still exists and works for systemctl

### Generated Report (if applicable)

ckb-next-dev-detect - Sun May 26 12:28:04 EDT 2024

```
Kernel: Linux fedora 6.8.9-301.fsync.fc40.x86_64 #1 SMP PREEMPT_DYNAMIC Sat May  4 16:18:13 UTC 2024 x86_64 GNU/Linux

OS Version:
LSB Version:	n/a
Distributor ID:	Fedora
Description:	Fedora Linux 40.20240524.0 (Bazzite)
Release:	40
Codename:	n/a

USB Devices:

Bus 005 Device 002: ID 1b1c:1b3f Corsair Corsair Gaming K68 Keyboard
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1b3f Corsair Gaming K68 Keyboard
  bcdDevice            3.12
  iManufacturer           1 Corsair
  iProduct                2 Corsair Gaming K68 Keyboard
  iSerial                 3 11034022AF2D84675AE48783F5001BC2
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0042
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     132
          Report Descriptors: 
[... 478 lines cut ...]
	libFLAC.so.12 => /lib64/libFLAC.so.12 (0x00007fd077fbe000)
	libvorbis.so.0 => /lib64/libvorbis.so.0 (0x00007fd077f8f000)
	libvorbisenc.so.2 => /lib64/libvorbisenc.so.2 (0x00007fd077ef7000)
	libopus.so.0 => /lib64/libopus.so.0 (0x00007fd077e8f000)
	libogg.so.0 => /lib64/libogg.so.0 (0x00007fd077e85000)
	libmpg123.so.0 => /lib64/libmpg123.so.0 (0x00007fd077e2b000)
	libmp3lame.so.0 => /lib64/libmp3lame.so.0 (0x00007fd077db4000)
	libbrotlidec.so.1 => /lib64/libbrotlidec.so.1 (0x00007fd077da6000)
	libnghttp2.so.14 => /lib64/libnghttp2.so.14 (0x00007fd077d7c000)
	libidn2.so.0 => /lib64/libidn2.so.0 (0x00007fd077d5a000)
	libgmodule-2.0.so.0 => /lib64/libgmodule-2.0.so.0 (0x00007fd07ab33000)
	libmount.so.1 => /lib64/libmount.so.1 (0x00007fd077d0a000)
	libbrotlicommon.so.1 => /lib64/libbrotlicommon.so.1 (0x00007fd077ce7000)
	libunistring.so.5 => /lib64/libunistring.so.5 (0x00007fd077b3a000)
	libblkid.so.1 => /lib64/libblkid.so.1 (0x00007fd077b00000)

Environment:
XDG_SESSION_DESKTOP=KDE
XDG_SESSION_TYPE=wayland
XDG_CURRENT_DESKTOP=KDE
WAYLAND_DISPLAY=wayland-0
QT_WAYLAND_RECONNECT=1
DISPLAY=:0
QT_AUTO_SCREEN_SCALE_FACTOR=0
```

gravityfargo, 2024-07-12:
Which command are you referring to? This one?
```bash
sudo ckb-next-daemon --enable-experimental
```
If you haven't already, try [editing the systemd service file](https://github.com/ckb-next/ckb-next/wiki/Linux-Installation#enabling-experimental-support-on-boot).

gravityfargo, 2024-12-11:
Excellent work, I've made modifications to the wiki after verifying.

closed 2024-12-11 by gravityfargo