ckb-next/ckb-next #1067: Requesting support for K70 MAX RGB

sunsh9876543210, 2024-06-09:
### What kind of device is it?

Keyboard

### What is the name of the device?

K70 MAX RGB Magnetic

### Link to Corsair's Product Page

https://www.corsair.com/lm/en/p/keyboards/ch-910961g-na/k70-max-rgb-magnetic-mechanical-gaming-keyboard-adjustable-corsair-mgx-switches-steel-grey-ch-910961g-na

### Generated Report

Drag your generated report here.

gravityfargo, 2024-07-12:
Please generate a report and attach it. I am closing this for now.

> In the ckb-next application, go to the Settings tab of the user interface and click the Generate Report button. Alternatively, run ckb-next-dev-detect in a terminal.

closed 2024-07-12 by gravityfargo

Melk0rr, 2024-08-19:
Hello, here is a report for this device
[ckb-next-report-1724078965.gz](https://github.com/user-attachments/files/16661860/ckb-next-report-1724078965.gz)