ckb-next/ckb-next #1074: Corsair K100 RGB Mechanical keyboard

rskunath, 2024-07-13:
### Describe your situation in detail.

My beloved K70 has finally succumbed to key chatter so I'm looking for a new keyboard. K70s are unobtanium, so I saw the K100 RGB Mechanical and that has the same MX-Speed switches my K70 does that I like. So, I was trying to find out if this model is supported and how supported it is? I don't need or want the optical switches, so going with the existing mechanicals that I like now. Are the wheel and the extra G keys supported? The SKU on Corsair's site of this K100 model is: CH-912A014-NA and I was hoping someone that has one might let me know about the support level. TIA.

### Additional Details

Add any additional details here.

### Generated Report (if applicable)

Drag your generated report here if it's applicable.

kimjongbing, 2024-07-23:
The keyboard and the G keys work for me. The wheel does not. If you slightly even touch the wheel, it will start typing out nonsense and trigger other keys. Not sure how to make the wheel do nothing.

gravityfargo, 2024-08-05:
@rskunath  According to[ the wiki](https://github.com/ckb-next/ckb-next/wiki/Supported-Hardware#keyboards), the K100 RGB Mechanical is supported, but we cannot be sure of that device if we do not know the product ID. Unfortunately, you can only find that out if you purchase it and plug it in. 

@kimjongbing Please submit a bug report.

closed 2024-08-05 by gravityfargo

rskunath, 2024-08-10:
That makes sense, but there were two reports that the wheel does not work. Since the K100 has the wheel instead of the 3 buttons the K70 has, this makes the thing somewhat unusable.

So if there is a wheel bug then it is not really fully supported.